dirkhh/adsb-feeder-image #10: 500 error for /backup and /restore

MattDahEpic, 2023-06-24:
Using the navbar buttons or manually navigating to `/backup` or `/restore` results in a 500 error with the following message logged in the web server logs:
```
Jun 23 20:01:25 adsb flask[858]:   File "/usr/local/lib/python3.9/dist-packages/jinja2/environment.py", line 936, in handle_exception
Jun 23 20:01:25 adsb flask[858]:     raise rewrite_traceback_stack(source=source)
Jun 23 20:01:25 adsb flask[858]:   File "/usr/local/share/adsb-pi-setup/templates/backup.html", line 2, in top-level template code
Jun 23 20:01:25 adsb flask[858]:     {% set active_page = "backup" %}
Jun 23 20:01:25 adsb flask[858]:   File "/usr/local/share/adsb-pi-setup/templates/base.html", line 34, in top-level template code
Jun 23 20:01:25 adsb flask[858]:     style="display: {% if env_values['BASE_CONFIG'] == "1" or
Jun 23 20:01:25 adsb flask[858]:   File "/usr/local/lib/python3.9/dist-packages/jinja2/environment.py", line 466, in getitem
Jun 23 20:01:25 adsb flask[858]:     return obj[argument]
Jun 23 20:01:25 adsb flask[858]: jinja2.exceptions.UndefinedError: 'env_values' is undefined
Jun 23 20:01:25 adsb flask[858]: 127.0.0.1 - - [24/Jun/2023 04:01:25] "GET /backup HTTP/1.1" 500 -
```
I'm not too familiar with Flask but it appears to be an error with showing/hiding the navbar in the base template. Changing the order of the conditions in the `if` on the affected lines to force a short-circuit in the condition on the affected pages does not resolve the issue.

Occurs on v0.9.2 and testbuild-g-f6bd5de6.

dirkhh, 2023-06-24:
Whenever a page is rendered, we need to pass in the env_vars so that the base template works.

I didn't think there was any caller left that didn't do that, but it looks like you found one.
I can't quite tell where the render_template() is called from. I'll poke at this...

dirkhh, 2023-06-24:
Crud.
I clearly didn't test backup and restore after the last edit of the base template.
All that is needed is to pass `env_values=ENV_VALUES.envs` to the `render_template()` calls for backup and restore.
Again, another great find!

commit 19271f3c (2023-06-24): Ensure backup and restore routes have env_values (resolves #10)
commit 3c0e621c (2023-06-24): Merge pull request #11 from MattDahEpic/backup-env_values

Ensure backup and restore routes have env_values (resolves #10)
closed 2023-06-24 by dirkhh