dirkhh/adsb-feeder-image #12: Configuration is wiped after Pi reboots

chevyman142000, 2023-06-30:
Just set this up last night with an old Raspberry Pi 3 I had laying around to test with. In doing some testing, I ran into an issue with the backup and restore pages which I see there was already an issue opened for. Additionally, any time I reboot my Pi (either reboot from ssh session or just pulling power) it loses the config and is basically like it is new out of the box. Let me know what I can provide to help troubleshoot! Thank you for the help!!

dirkhh, 2023-06-30:
Obviously not the intended behavior :)

Hmm. When you look at `/opt/adsb/.env` do you see your settings in that file, or is that replaced with a clean file?
What does `systemctl status adsb-setup.service` tell you? Is the service running?

chevyman142000, 2023-06-30:
haha, I figured as much! Thanks for the quick response on this. Here is a screenshot showing both. 

[image]

dirkhh, 2023-06-30:
`BASE_CONFIG=0` implies that configuration never completed... that's not a good thing.
Can you tell me about the workflow that got you to this point?
You wrote the image to an SD card, booted, then what did you do?

chevyman142000, 2023-06-30:
For sure! I downloaded the image, wrote it to an SD card, popped it in, waited for it to fully boot, went to the website and then set the lat/lon/height/etc and selected pick aggregators and hit submit. I should have mentioned, my screenshots are from right after I rebooted it and it was wiped. I'm including another screenshot of the same info after I set all of the settings again, just to see if there is a difference.

[image]

dirkhh, 2023-06-30:
So here `BASE_CONFIG` is set. I need to figure out how it gets erased...
That certainly gives me something to chase down.

I am traveling until mid next week which means I can't try to recreate the issues, but maybe this can be solved just by staring at the code...

chevyman142000, 2023-06-30:
Sounds great. Let me know if there is anything else I can grab that might be helpful. I'm just in a testing state right now...waiting on the UAT978 stuff and the backup/restore to be fixed before I move my production setup over to this.

dirkhh, 2023-06-30:
Which version of the image are you using... UAT978 is supported and if it detects that you have more than one SDR connected it should automatically ask you to assign them to their different functions...
Unfortunately backup/restore is indeed broken on Raspberry Pis right now (it works fine on the other board... the Pi is running an older base OS with an older Python version)

dirkhh, 2023-06-30:
in the meantime, I'm pretty sure I understand what's going wrong with the reboot and believe that I have a fix for that. But since I don't have any hardware with me for testing, I can't really continue to work on this until mid next week.

chevyman142000, 2023-07-01:
Oh great, I didn't realize UAT978 was already implemented. I'll plan to move over then as soon as my issue and the backup/restore ones are fixed. Thanks again and let me know as soon as you have a new version I can test out. Thanks again for the help and being so responsive!

dirkhh, 2023-07-01:
I won't be back home until mid next week and I want to make sure I more carefully take this through its paces on a Pi... So I'm guessing this should be in decent shape in about a week.

commit 8363890f (2023-07-01): disable adsb-init after things are set up

And just in case that fails, also don't overwrite an existing .env file after
reboot.

Fixes: #12
closed 2023-07-04 by dirkhh