dirkhh/adsb-feeder-image #15: RadarViertuel Config Not Working

chevyman142000, 2023-07-05:
I'm slowly setting up my Pi to replace my current setup and went to activate my RadarVirtual feeder key. I pasted it in and submitted my changes, but it doesn't appear to properly save the key. I also checked the log files and have included a screenshot of what I'm seeing below. Let me know if there is anything else I can grab to help troubleshoot. Sorry for all of the issues! :-) 

[image]

dirkhh, 2023-07-06:
I'll try to reproduce this here. But in the meantime, can you check in your `.env` file the values for `RV` and `FEEDER_RV_FEEDER_KEY` ? Do they look right?

dirkhh, 2023-07-06:
Never mind - I found the bug. And it was an embarrassing one for sure. `-` vs `_` typo that prevented the information from being written to the `.env` file :(

chevyman142000, 2023-07-06:
Haha, I hate when that happens! Thanks for digging into it so quickly!

commit 503d5079 (2023-07-06): fix Radar Virtual setup

This appears to be a feeder no one ever tested.

Fixes: #15
closed 2023-07-07 by dirkhh