dirkhh/adsb-feeder-image #16: ZeroTier Config Not Working

chevyman142000, 2023-07-07:
Found another one for you! I signed up for a ZeroTier account, never used it before, and added my network id in. When I hit submit, I recieved an Internal Server Error page, screenshot below. I checked the .env file and it looks like it saved the id, but it doesn't appear to be working. I have rebooted and I also noticed that the ID no longer shows up in the web ui. Let me know if I can grab any other info!

[image]

dirkhh, 2023-07-07:
Thanks. Found. Fixed.

closed 2023-07-07 by dirkhh

chevyman142000, 2023-07-07:
Yes, this does appear to be working now. However, after you submit, the field is blank and doesn't show the ID anymore. Not sure if that is by design or not.

commit 95e173a8 (2023-07-07): fix call to ensure zerotier is enabled

Fixes: #16