dirkhh/adsb-feeder-image #20: PiAware Errors

chevyman142000, 2023-07-07:
Around 30 minutes of running, I noticed that FlightAware was no longer receiving data from my feeder. I'm seeing some errors in the container logs, see screenshot below. I also see the container marked as unhealthy if I do a docker ps. I also included a screenshot of the PiAware status page.

[image]

[image]

dirkhh, 2023-07-07:
That's a strange one. Hmm. I can't quite figure out what's going on there.
Right now I reboot my test-rig so often that I guess I have never run into that one.

chevyman142000, 2023-07-07:
Looks like it is consuming quite a bit of CPU too. Even on 0.9.4 I'm having the same issue. Here is a screnshot of htop. The PiAware status page looks the same as the above screenshot too, even though it appears to be still running.

[image]

chevyman142000, 2023-07-09:
Just another follow up now that things are working better. It seems after some time of running the PiAware container starts to consume a lot of CPU and I'm wondering if when that happens it eventually locks/crashes the Pi? Here is what I'm seeing in the log, looks like the log stopped here even though this is from about 2 hours ago. Also see a screenshot of the CPU graph. About the same time the log stops the CPU spikes and the PiAware container is using 100% of a core.

[image]
[image]
[image]

chevyman142000, 2023-07-09:
One more screenshot showing the CPU graph after I restarted the PiAware container. 
[image]

dirkhh, 2023-07-09:
Wow. That's wild. Mine has been running now for 20+ hours and seems just fine.
How are you connecting to the network. WiFi? Ethernet?

dirkhh, 2023-07-19:
again, not a feeder image issue

closed 2023-07-19 by dirkhh