dirkhh/adsb-feeder-image #22: Feature Request: Dark Mode

chevyman142000, 2023-07-07:
This is just a nice to have and nothing is broken. I just wanted to get it listed here, in case it wasn't already on your roadmap/plans. Would love to see a dark mode similar to how the graphs have it enabled.

dirkhh, 2023-07-07:
I'll admit that this is slightly lower on my list of priorities compared to "it works"... :)

chevyman142000, 2023-07-07:
Absolutely understood, no problem at all! I agree, other stuff is more important.

katlol, 2023-07-08:
I recommend [dark reader](https://addons.mozilla.org/en-US/firefox/addon/darkreader/), then you can darkify anything: 
[image]

dirkhh, 2023-08-04:
I moved this over into the wiki so I can clean up the issues here...
https://github.com/dirkhh/adsb-feeder-image/wiki/Feature-requests

closed 2023-08-04 by dirkhh