dirkhh/adsb-feeder-image #24: PiAware MLAT Issues

chevyman142000, 2023-07-07:
When I check the PiAware status page, I see MLAT is in an orange state and says Local Clock Source Unstable. I also checked the container logs and see the following error multiple times in the log with varying percentages.  NOTICE from adept server: 18% of multilateration messages (UDP) are not reaching the server - check your network?

[image]

[image]

dirkhh, 2023-07-09:
I finally switched my "real" receiver out for an adsb feeder image based one... so I am officially eating my own dog food :)
Here's the screenshot of my status on the FlightAware website... So it seems mlat is working just fine, here.
It looks like there are some odd network issues in that log as well. I'm not sure what to tell you - obviously (well, at least to me) the feeder image is simply re-using the SDR-Enthusiast docker images... so I don't think this should be something specific to the way I install them.
I just pushed a commit that adds UAT978 support for Piaware/RadarBox, but that shouldn't have anything to do with what you are seeing here. So I'm not sure what to poke at, next.
[image]

dirkhh, 2023-07-19:
I think we resolved that this isn't an issue with the Feeder Image as the same happened with a direct docker install...

closed 2023-07-19 by dirkhh