dirkhh/adsb-feeder-image #29: 0.9.5 Not Working

chevyman142000, 2023-07-08:
Re-imaged with 0.9.5 and setup manually, did not restore backup. Only detecting 1 SDR, not both that were plugged in upon initial boot. Tried to edit the .env file to set it to 2, but upon reboot it set it back to 1. Unplugged my 978 sdr and only left the 1 plugged in and it still isnt' working. Dump1090 maps don't show anything. Here is the log from the Ultrafeeder container, something is unhappy.

[image]

dirkhh, 2023-07-08:
Yeah - as I said on Zulip, I think the problem may be that you have an SDR with a different USB Id that the image doesn't recognize. And because of the way I changed things around in the multi-SDR detection, it now doesn't find it anymore.
@chevyman142000 - can you run `lsusb` on the Pi and attach the output here? Getting your specific model added should be super simple. But the bigger question of course is... are there others?

@wiedehopf - do you happen to know if there's a list of all of the USB Ids of RLT SDRs out there somewhere?

dirkhh, 2023-07-08:
So I found [this list](https://www.reddit.com/r/RTLSDR/wiki/compatibility/) - wow, that's quite a few 🤯
Again, @wiedehopf - any additional insights?

katlol, 2023-07-09:
FWIW; AFAIK the E4000 are deaf at around 1090MHz

dirkhh, 2023-07-09:
I didn't add the E4000 - just the alternative RTL PID (2832 vs 2838)
Given that this works now, closing this one.

commit 020cfc47 (2023-07-09): recognize additional RTL-SDR compatible dongles

Not sure if this list is complete, but it's a start.

Fixes: #29
commit ced41774 (2023-07-09): recognize additional RTL-SDR compatible dongles

Not sure if this list is complete, but it's a start.

Fixes: #29
commit 19f661ab (2023-07-09): recognize additional RTL-SDR compatible dongles

Not sure if this list is complete, but it's a start.

Fixes: #29
closed 2023-07-09 by dirkhh