dirkhh/adsb-feeder-image #35: Typo in tar1090 config.js when adsb.lol Route API enabled

Ottergoose, 2023-08-04:
In 0.11.0,  when adsb.lol route api is enabled, config.js is populated with `useRouteAPI = True;`, which throws an error as "True" is undefined. Should be  'useRouteAPI = true;` (lowercase)

dirkhh, 2023-08-04:
I tried this on so many different boards.
With so many different configurations.
I have no idea how that slipped past me.

I'll fix this and release a new version tomorrow.

/D

dirkhh, 2023-08-04:
root caused, fix is going through testing right now.
it's so frustrating that I once again managed to break something as obvious as this... but yeah, 0.11.1 will be out in the next few hours
thanks for the report!!!

Ottergoose, 2023-08-04:
Happy to help! Love this software, def the easiest way to feed all of the aggregators I want to feed, and it's so fast/easy to start over using the Raspberry Pi imager that remembers to do the network configuration. Keep up the good work!

dirkhh, 2023-08-04:
@Ottergoose would you be able to give this test build a quick spin?
https://github.com/dirkhh/adsb-feeder-image/releases/download/latest/adsb-feeder-raspberrypi64-g-64b2a170.img.xz

Ottergoose, 2023-08-04:
Can do, will report back in a few minutes

dirkhh, 2023-08-04:
> Can do, will report back in a few minutes

never mind... I can still reproduce it with a fresh install.
my fix fixed it when rsynced to the Pi and run there, but on a fresh install I once again ended up with True in the config.js

back to the drawing board :(

Ottergoose, 2023-08-04:
D'oh!

dirkhh, 2023-08-04:
new image to test. This time I mean it. I mean. Err. This time I tested this very image (and not just its equivalent local build)... 🤣

https://github.com/dirkhh/adsb-feeder-image/releases/download/latest/adsb-feeder-raspberrypi64-g-9f89db76.img.xz

Ottergoose, 2023-08-04:
I'll give it a whirl right now, thank you!

closed 2023-08-04 by dirkhh

Ottergoose, 2023-08-04:
Confirmed resolved per build you linked to most recently; thank you!

dirkhh, 2023-08-04:
Thanks again for reporting and testing the fix 

0.11.1 will be announced shortly...

commit 08b943ea (2023-08-04): don't use Python True/False for javascrip values

JS fails on a True or False as constants, it requires true/false or 1/0.

Fixes #35