dirkhh/adsb-feeder-image #40: Reformat Feeder Homepage with table of aggregators + status + links

Ottergoose, 2023-08-11:
Below extant mapping interface and administration links, it'd be slick to have a table listing aggregators, status, and various links for the aggregators being fed.

- ADSB status
- 978 status
- MLAT status
- MLAT peer count
- Link to service's status page(s)
- Link to your map on that service

A few of those links, for reference

ADSB.lol
- https://api.adsb.lol/api/0/me
- https://status.adsb.lol/
- https://my.adsb.lol/

ADSB.fi
- https://api.adsb.fi/v1/myip

ADSB.one
- https://api.adsb.one/feed-status

ADSBx
- https://www.adsbexchange.com/myip/

The Air Traffic
- https://theairtraffic.com/iapi/ip_stats.json/
- https://theairtraffic.com/feed/myip/

FlightRadar24
- :8754/monitor.json
- :8754

dirkhh, 2023-08-11:
Cool idea. I like it

katlol, 2023-08-11:
https://mlat.adsb.lol with note that if privacy=on you are not there

dirkhh, 2023-08-12:
A good chunk of this is implemented and available in testbuilds: https://github.com/dirkhh/adsb-feeder-image/releases/tag/latest

Ottergoose, 2023-08-13:
Nice! I'm hoping for a dashboard that parses those JSON feeds from the various feeders, but this is a great start.

Once things slow down at my day job, this might be a fun project for me to tackle and try to learn some new front-end stuff...

dirkhh, 2023-08-13:
No argument that this would be a great next step...
I'd love to get a PR (I'm busy trying to get the DietPi based images to a point where I'm comfortable releasing them)

Ottergoose, 2023-08-13:
I'll see what I can do; good luck with the Diet Pi updates, and thanks again for the ongoing efforts.

dirkhh, 2023-09-10:
I know that this is far from what you are looking for, but v0.15.0 has the first baby step of what I think you had in mind.
Let me know what you think 🙃

Ottergoose, 2023-09-11:
A fine first step, thank you!

One minor issue - flyover.com needs to be changed to flyover.io in the new list/table

Ottergoose, 2023-09-18:
Closing, looks great in 0.16.0, thank you!

closed 2023-09-18 by Ottergoose