dirkhh/adsb-feeder-image #41: Feature Request: support flyovr.io and radarplane.com

Ottergoose, 2023-08-14:
I don't know anything about these folks, but looks like another couple of indie sites to feed:

- https://www.flyovr.io/
- https://radarplane.com/feed

dirkhh, 2023-08-14:
flyovr not only lacks a vowel, they also lack a privacy policy.
Slick site, but... not really giving me the warm and fuzzy.
radarplane has a privacy policy that could be summarized as "what's yours is mine, and what's mine is none of your business". Not thrilled by it. 

Both seem like just adding more noise to an already overcrowded space. I'm not convinced that either is useful or will create value. But then (devil's advocate) I'm not so sure about this regarding a couple of the ones that I have already listed... 🤷🏼‍♂️

I'm somewhat undecided. I guess I should just add them and the heck with it.

Ottergoose, 2023-08-14:
inclusion != endorsement; being easy and platform agnostic is what makes this software so useful

dirkhh, 2023-08-14:
You are so correct. Thank you for reminding me... sometime it's easy to get lost in the details.

https://github.com/dirkhh/adsb-feeder-image/commit/6f95045a017ade8c69281fc5af2939a4985975bd

dirkhh, 2023-08-14:
in my testing something isn't going right with the new aggregators.
I need to walk away and try again tomorrow. Brain is flat

closed 2023-08-14 by dirkhh