dirkhh/adsb-feeder-image #44: List supported aggregators on the project homepage

Ottergoose, 2023-08-15:
IMHO, one of the biggest selling points of this software is ease of setup, and ability to feed multiple aggregators.

Perhaps the introduction could be amended, `Making it easy to collect ADS-B data and feed it to multiple aggregators.`

An additional bullet point could be added with a list of supported aggregators? 

- Feed your ADS-B data to [adsb.fi](https://adsb.fi), [adsb.lol](https://adsb.lol), [ADS-B Exchange](https://adsbexchange.com), [ADSBHub](https://adsbhub.org), [ADS-B One](https://adsb.one), [AVDelphi](https://www.avdelphi.com), [FlightAware](https://flightaware.com), [FlightRadar24](https://www.flightradar24.com), [Fly Italy Adsb](https://flyitalyadsb.com), [FLYOVR.io](https://www.flyovr.io/), [OpenSky Network](https://opensky-network.org), [Plane.watch](https://plane.watch), [Plane Finder](https://planefinder.net), [Planespotters.net](http://planespotters.net), [RadarBox](https://www.radarbox.com), [RadarPlane](https://radarplane.com/), [Radar Virtuel](https://www.radarvirtuel.com), [TheAirTraffic](http://theairtraffic.com), and more!

dirkhh, 2023-08-15:
The trick is to find the balance between "short and easy to read" and "cover everything that's important" 🤷🏼‍♂️

I do like that list.

Again, where are you thinking this should be listed. I assume adsb.im ?

Ottergoose, 2023-08-15:
Right, but, I suspect most people looking for this software are looking for a solution "I want to feed data to X, Y and Z," listing them on the homepage ( adsb.im/home ) will help them discover this software.

Here is list again, with open data commitment providers up front:

- Feed your ADS-B data to [adsb.lol](https://adsb.lol/), [Fly Italy Adsb](https://flyitalyadsb.com/), [TheAirTraffic](http://theairtraffic.com/), [adsb.fi](https://adsb.fi/), [ADS-B Exchange](https://adsbexchange.com/), [ADSBHub](https://adsbhub.org/), [ADS-B One](https://adsb.one/), [AVDelphi](https://www.avdelphi.com/), [FlightAware](https://flightaware.com/), [FlightRadar24](https://www.flightradar24.com/), [FLYOVR.io](https://www.flyovr.io/), [OpenSky Network](https://opensky-network.org/), [Plane.watch](https://plane.watch/), [Plane Finder](https://planefinder.net/), [Planespotters.net](http://planespotters.net/), [RadarBox](https://www.radarbox.com/), [RadarPlane](https://radarplane.com/), [Radar Virtuel](https://www.radarvirtuel.com/), and more!

dirkhh, 2023-08-15:
I will update the home page. I'll leave this issue open so I don't forget 😎

dirkhh, 2023-08-15:
I think I achieved this: https://adsb.im/home

dirkhh, 2023-08-15:
Thank you so much for not only the suggestion, but for also collecting the URLs and making it easy to use your code...
the page is actually HTML (long story), but thanks to https://codebeautify.org/markdown-to-html that's a trivial conversion :)

closed 2023-08-15 by dirkhh

Ottergoose, 2023-08-15:
You're welcome, I'm glad I'm able to contribute. Thanks for all of your time/effort thrown into this project, it's much appreciated.