dirkhh/adsb-feeder-image #50: Feature Request: Add Dozzle for easy viewing of container logs.

seanford, 2023-08-25:
Came across Dozzle on the Reddit /selfhosted forum.

Took me 5 minutes to implement on my adsb-feeder-image Pi4.  

Runs on Port 8080, so I moved that to 9999.

I think this is would be great addition for this project.  It helped me realize I had entered my radarvirtuel key wrong 2 minutes after installing it.  Also let you see which services are consuming resources.

More info at: https://dozzle.dev/

dirkhh, 2023-08-25:
This does look interesting... admittedly my far cruder current version should be enough to get the job done, but this would be a better user experience.

My concern is adding dependencies, and adding overhead (especially on low memory boards).
But I'll play with it and see if this fits the design goals.

Thanks for the pointer!

seanford, 2023-08-25:
Understood.  Maybe make it an optional install, just like the feeders.

Doesn't seem to be adding much overhead to my setup, which is a Pi4 / 2GB

dirkhh, 2023-08-25:
It's really the 512M boards that I worry about

dirkhh, 2023-08-25:
oh. oh my.
I'm in love.
Ok, this will be a feature in the next version.
Thank you SO MUCH for suggesting it!

seanford, 2023-08-25:
Glad you like it....I literally came across it by mistake and immediately thought it would be perfect for this.  Took me all of 5-10 minutes to read up and install it manually, but I'm glad it will be integrated going forward.

commit 941cd1bc (2023-08-26): enable Dozzle for logging

The simplistic dumping of potentially massive text simply wasn't nearly as
useful. And Dozzle uses surprisingly little memory and next to no CPU.

Fixes: #50
closed 2023-08-28 by dirkhh