dirkhh/adsb-feeder-image #53: Bug: connection refused to :9999 for logs

Ottergoose, 2023-08-28:
v0.14.2(main)-230827.11 with containers from 2023-08-19T17:38:37-05:00, running on Raspberry Pi 3 Model B Plus Rev 1.3. adsb-feeder-raspberrypi64-v0.12.0.img

I know this is new functionality, but when I click on the Logs link to :9999, I get an NS_ERROR_CONNECTION_REFUSED in Firefox and ERR_CONNECTION_REFUSED from Chrome.

Thanks!

dirkhh, 2023-08-28:
Interesting. Did you install from scratch or did you upgrade?

dirkhh, 2023-08-28:
Oh wait, I can see that you upgraded.

Hmm. Depending on which version of the feeder image you started from, there's a chance that you are missing a container restart.

Easiest solution would be a reboot, I guess

Ottergoose, 2023-08-28:
Updated from 0.12 (I think), resolved with a reboot. Thanks!

dirkhh, 2023-08-28:
excellent. closing.