dirkhh/adsb-feeder-image #57: Feature request: enable MLAT on Flightradar24

andreagdipaolo, 2023-09-25:
I noticed that MLAT was disabled by default.
I enabled it in its own config, because I saw nowhere in the ADSB.im config a toggle to enable it.
It works but it still shows as disabled in the feeding table.

P.S. a key to understand the symbols on the table would be great.
Awsome project, thanks for your work!

andreagdipaolo, 2023-09-26:
Thanks for replying.
If I don't recall wrong, they suggested to turn MLAT off because they were concerned of poor results by weak systems, however there was nothing wrong with continuing to feed MLAT with good quality data.
I believe that should be an option, but it's just my suggestion. Anyway, detection of MLAT being active doesn't work.

Also, Radarbox has MLAT active and it's not detected in the table either.

dirkhh, 2023-09-26:
I'm not 100% sure I understand what you are asking here...
Not all aggregators give back the information I want to present. E.g., Radarbox doesn't provide me with an API to do so.
The `.` in the column says "I can't tell". Only a `-` indicates that we know that MLAT isn't working.
Therefore, only if you have a `-` in either of those columns will the checkmark turn red (and only if both are `+` will it be green. In the cases where we just don't know, it stays black.

andreagdipaolo, 2023-09-26:
> I'm not 100% sure I understand what you are asking here...

Simply being able to control the basic feeder settings directly via the aggregator. But maybe I missed the point of the project and it's not supposed to be like that? A way to easyly edit each aggregator's config via your "hub".

> Not all aggregators give back the information I want to present. E.g., Radarbox doesn't provide me with an API to do so.

Oh, okay, didn't know that. Thanks!

> The . in the column says "I can't tell". Only a - indicates that we know that MLAT isn't working.
Therefore, only if you have a - in either of those columns will the checkmark turn red (and only if both are + will it be green. In the cases where we just don't know, it stays black.

Wouldn't it be better to have a legend/key somewhere? Maybe I missed it.

dirkhh, 2023-09-26:
There will be a legend in the next version.
That is a great idea.

Adding more features and capabilities is a trade off... the target audience is people who aren't very technical. So making things more complex isn't in the best interest of the target audience. The system is already far more complex that I had intended it to be.

That said, I can see a way to bridge the gap by adding more options and telling the technical user to edit the `.env` file...

I'll need to figure out how many things should become configurable, how to make sure this doesn't break the "simple" use case. And how to document it.

closed 2023-10-04 by dirkhh

dirkhh, 2023-10-15:
I realized that I never followed up here - but I assume you found the change that I made that allows you to set `FEEDER_FR24_MLAT` to 1 which should enable MLAT in the FlightRadar24 docker container. I'll admit that this hasn't received a lot of testing, but when I ran it for a little while it seemed to work.

andreagdipaolo, 2023-10-15:
> I realized that I never followed up here - but I assume you found the change that I made that allows you to set `FEEDER_FR24_MLAT` to 1 which should enable MLAT in the FlightRadar24 docker [...]

Thanks, but where should I put that string? In which config file?

dirkhh, 2023-10-16:
`/opt/are/config/.env`

andreagdipaolo, 2023-10-16:
> `/opt/adsb/config/.env`

This file is empty, is that right?

andreagdipaolo, 2023-10-16:
The setting seems to be enabled correctly by opening the web config (just for checking, haven't edited it) but in the main page of Fr24feed it says N/A and in adsb.im webui MLAT has a dot.

[image]

[image]

commit 2bd016cf (2023-10-04): allow manual enabling of MLAT on FlightRadar24

With this change a user who wants to enable MLAT can do that through their .env
file.

Fixes #57