dirkhh/adsb-feeder-image #62: Can't upgrade form 0.17.2 to 0.17.4

ramphex, 2023-10-04:
As the title says, when I attempt to upgrade to 0.17.4 from 0.17.2 from the Expert page, it tells me "please wait, this could take several minutes" and then I end up on the main page and it still shows as running 0.17.2 with 0.17.4 available. I have tried Update Stable and Update Beta with the same result.

dirkhh, 2023-10-04:
There's a bug in v0.17.1 that stayed undetected until v0.17.4 - as a result v0.17.1/2/3 won't upgrade from the web UI.
The workaround is fairly simple (and a one-time thing):
- on the expert page set an ssh key for the root account.
- ssh as root into the feeder using that key
- run `/opt/adsb/feeder-update stable`

Once that completes, you are back to normal and future updates once again will work from the web UI.

ramphex, 2023-10-04:
Thanks, running the command got the update to the most recent 0.17.5