dirkhh/adsb-feeder-image #65: Excessive disk writes

andreagdipaolo, 2023-10-15:
I've noticed that since I installed this software on my Rock Pi 4 via DietPi the write count on my disk has increased hugely.
I have a constant 130 kB/s write speed with 14 IOPS.
Looking at htop the main causes seem to be `fr24feed-log` with something like 20 kb/s constant and especially `collectd` (inside ultrafeeder).
Is it normal? Can it be optimized to store data in RAM and write to the disk less often and less in general?
I have an NVMe so that's less concerning, but if you run the system on a microsd this would kill it very fast.
Thanks

dirkhh, 2023-10-15:
This is designed to be writing to a tmpfs. So no, this isn't normal at all.
As intended, the containers all should write to subdirectories of the hosts `/run` filesystem, which should be a tmpfs.
From what you are saying, it sounds like that isn't how it is working out for you - and I'd love to understand why so I can fix that :)

I assume you installed the feeder via `dietpi-software`?
What do you get when run `mount | grep "^/run "` (there's a space between the `n` and the `"`)

andreagdipaolo, 2023-10-15:
> I assume you installed the feeder via dietpi-software?

Yep, I'm the guy from the other issue on the dietpi repo 😆

> What do you get when run mount | grep "^/run " (there's a space between the n and the ")

Nothing

dirkhh, 2023-10-15:
and that is the problem in a nutshell. I was under the impression that DietPi always establishes `/run` as tmpfs, but that doesn't seem to be the case for you.
Just to confirm this, can you run `df -h /run` -- that will likely give you the SDcard device that holds your root fs. Which isn't great. I don't have a Rock Pi 4... what are the specs? How much memory does it have? I'm just surprised that this wouldn't have a tmpfs as `/run`...

andreagdipaolo, 2023-10-15:
> I don't have a Rock Pi 4... what are the specs? How much memory does it have? I'm just surprised that this wouldn't have a tmpfs as `/run`...

```
       _,met$$$$$gg.          dietpi@DietPi 
    ,g$$$$$$$$$$$$$$$P.       ------------- 
  ,g$$P"     """Y$$.".        OS: Debian GNU/Linux 12 (bookworm) aarch64 
 ,$$P'              `$$$.     Host: Radxa ROCK Pi 4B 
',$$P       ,ggs.     `$$b:   Kernel: 6.1.50-current-rockchip64 
`d$$'     ,$P"'   .    $$$    Uptime: 9 days, 19 hours, 25 mins 
 $$P      d$'     ,    $$P    Packages: 536 (dpkg) 
 $$:      $$.   -    ,d$$'    Shell: bash 5.2.15 
 $$;      Y$b._   _,d$P'      Terminal: /dev/pts/1 
 Y$$.    `.`"Y$$$$P"'         CPU: (6) @ 1.512GHz 
 `$$b      "-.__              Memory: 1908MiB / 3863MiB 
  `Y$$
   `Y$$.                                              
     `$$b.                                            
       `Y$$b.
          `"Y$b._
              `"""

```
I'm running on NVMe and I have no microsd installed, bootloader via SPI.

> and that is the problem in a nutshell. I was under the impression that DietPi always establishes `/run` as tmpfs, but that doesn't seem to be the case for you. Just to confirm this, can you run `df -h /run` -- that will likely give you the SDcard device that holds your root fs. Which isn't great. 

```
Filesystem      Size  Used Avail Use% Mounted on
tmpfs           387M   48M  339M  13% /run
```

dirkhh, 2023-10-15:
hm.
so it is on tmpfs
not sure why the mount command above didn't catch that, but whatever.
But then it makes no sense why you would see disk writes.

I'm starting to run out of ideas. How about this

`docker exec -it ultrafeeder bash`
that should give you a shell inside the container
`mount | grep run`
that should show that `/run` (and `/run/airspy` - which you only use if you have an airspy SDR) are on tmpfs

All the other filesystems should really receive not a lot of writes.

I have a few images running and am trying to figure out how to reproduce the behavior that you are seeing. None of them show particularly high IO / write rates...

dirkhh, 2023-10-15:
Ahh, I think I know what it is. The way Ultrafeeder writes the graphs has them end up in `/opt/adsb/config/ultrafeeder` -- and that might be the culprit. Weird that I didn't notice this on my long-running images. I'll need to figure out how to either make that tunable or move it to a tmpfs as well.
Thanks for the report!

andreagdipaolo, 2023-10-15:
> Ahh, I think I know what it is. The way Ultrafeeder writes the graphs has them end up in `/opt/adsb/config/ultrafeeder` -- and that might be the culprit. Weird that I didn't notice this on my [...]

Thanks, I hope you found the real cause. I've already burned 500GB this month 🙈
So it is the graphs pictures writing to the disk every few seconds?
I'm sure you'll be able to solve that. Thanks!

dirkhh, 2023-10-15:
ok... here's something to try:
edit `/opt/adsb/config/docker-compose.yml` (these edits will be overwritten by the next update... but if this works for you I'll add this to the next beta). Add environment variable that enables `REDUCE_IO` mode:
```
    environment:
        # ------
        # general parameters:
        - LOGLEVEL=error
        - TZ=${FEEDER_TZ}
        - GRAPHS1090_REDUCE_IO=true
  ...
```

dirkhh, 2023-10-15:
oh, after this you have to run `/opt/adsb/docker-compose-start` in order for the new setting to take effect.

If this ended up damaging your SD card, send me your address and I'll send you a replacement... 🤷🏼‍♂️

andreagdipaolo, 2023-10-15:
> oh, after this you have to run `/opt/adsb/docker-compose-start` in order for the new setting to take effect.
> 
> If this ended up damaging your SD card, send me your address and I'll send you a replacement... 🤷🏼‍♂️

That's so kind of you, but it's not your responsibility. You're not liable for the issues in your software.
Moreover, as I said, I'm not running on a microsd but on on a 2TB NVMe with a 600 TBW warranty, so 500GB is not a huge problem, it's like 1/1200 of the rating.
Thanks anyway!

Regarding the solution, I have added the `- GRAPHS1090_REDUCE_IO=true` string to the docker compose but if I launch your command it says `docker-compose-adsb is already running`. It launched successfully with sudo.

dirkhh, 2023-10-15:
yes, this requires a `sudo` - I should have mentioned that. Did this result in a reduction of disk IO?

andreagdipaolo, 2023-10-15:
> yes, this requires a `sudo` - I should have mentioned that. Did this result in a reduction of disk IO?

I was waiting for some samples. It seems like it's around 20% less bandwidth. Not a great improvement. 
I will be shutting down every docker container from other docker compose files and try only with the airplanes for better results.

andreagdipaolo, 2023-10-15:
After shutting down every other container (home assistant, immich, zigbee2mqtt, etc) it only reduced by another 20 kB/s settling at around 80 kB/s. With those containers active it was around 100, and they are quite active with all my zigbee sensors etc, so it's still an issue.

[image]

dirkhh, 2023-10-16:
Well, you can turn off writing the history completely. I have it running here and it does not create 500G a month, though.

Depending on whether you have a UAT978 container, or a Piaware container, there appear to be a couple more opportunities to reduce IO.
When you run `iotop -a` for an hour or so - which are the biggest culprits for writing data?

andreagdipaolo, 2023-10-16:
I have enabled all websites. From htop the one that seems to write more is fr24feed but if I didnt misunderstant it writes to tmpfs.

I am in Europe so I only run 1090.

I will run that and let you know, thanks!

andreagdipaolo, 2023-10-16:
Results are inconsistent.

```
    TID  PRIO  USER     DISK READ DISK WRITE>    COMMAND                                                                                                                                                                                                                
   3309 be/4 root          0.00 B     36.94 M s6-supervise fr24feed-log
    318 be/3 root          0.00 B     24.65 M [jbd2/nvme0n1p1-8]
2905594 be/4 70            0.00 B      9.36 M postgres: checkpointer
2905750 be/4 70          420.00 K      7.84 M postgres: homeassistant homeassistant_db 172.18.0.1(58326) idle
2904372 be/4 70            0.00 B      2.81 M postgres: stats collector
2868339 be/4 root          0.00 B      2.59 M readsb --net --quiet --lat bla bla bla
2905596 be/4 70            0.00 B      2.09 M postgres: walwriter
2905598 be/4 70            0.00 B   1600.00 K postgres: stats collector
   5290 be/4 root          0.00 B    996.00 K fr24feed
2905455 be/4 1883          0.00 B    368.00 K mosquitto -c /mosquitto/config/mosquitto.conf
2905717 be/4 root          0.00 B    336.00 K python3 -m homeassistant --config /config
 823393 be/4 root          0.00 B     96.00 K dockerd -H fd:// --containerd=/run/containerd/containerd.sock
2904147 be/4 root          0.00 B     84.00 K typesense-server
 827744 be/4 root          0.00 B     64.00 K dockerd -H fd:// --containerd=/run/containerd/containerd.sock
2905723 be/4 root          0.00 B     64.00 K python3 -m homeassistant --config /config
2905725 be/4 root          0.00 B     64.00 K python3 -m homeassistant --config /config
2905821 be/4 root          0.00 B     64.00 K python3 -m homeassistant --config /config
2905502 be/4 root          0.00 B     56.00 K dockerd -H fd:// --containerd=/run/containerd/containerd.sock
    784 be/4 root          0.00 B     44.00 K dockerd -H fd:// --containerd=/run/containerd/containerd.sock
 823395 be/4 root          0.00 B     44.00 K dockerd -H fd:// --containerd=/run/containerd/containerd.sock
 823426 be/4 root          0.00 B     44.00 K dockerd -H fd:// --containerd=/run/containerd/containerd.sock
    717 be/4 root          0.00 B     40.00 K dockerd -H fd:// --containerd=/run/containerd/containerd.sock
2905536 be/4 root          0.00 B     32.00 K node index.js
   3089 be/4 root          0.00 B      4.00 K rbfeeder_armhf
2905719 be/4 root          0.00 B      4.00 K python3 -m homeassistant --config /config
2905819 be/4 root          0.00 B      4.00 K python3 -m homeassistant --config /config
2905820 be/4 root          0.00 B      4.00 K python3 -m homeassistant --config /config
```

But with an average of 100 kB/s that should be wayyyy more.

[image]

Also by checking the SMART data of the disk I can clearly see that writes increase more than what the iotop says.

```
Data Units Read:                    449,698 [230 GB]
Data Units Written:                 2,173,672 [1.11 TB]
```

```
Data Units Read:                    449,699 [230 GB]
Data Units Written:                 2,173,819 [1.11 TB]
```

Data units are blocks of 512K so it's more or less 70MB

dirkhh, 2023-10-16:
I will fix the fr24 one, that seems to be the main culprit.
And I will continue to track this here as well

dirkhh, 2023-10-16:
Oh, and I cannot reproduce even remotely close to the numbers that you see... It's working more than it should but not nearly that much for me.

andreagdipaolo, 2023-10-16:
> Oh, and I cannot reproduce even remotely close to the numbers that you see... It's working more than it should but not nearly that much for me.

I don't know what could be different between us...
I am using a Flightradar green stick with autogain and I believe it's a quite common setup.
Fr24 logs in the webui of the feeder are very short (one line every 5 seconds), but probably they are two different logs, I suppose.

dirkhh, 2023-10-16:
For me we are writing about 1MB a minute (WAY too much), so 1.6G a day - which is just ridiculous.
I'll fix this for the next version.
Thanks for your patience.

dirkhh, 2023-10-16:
Ok, here is what should be a temporary fix to the FlightRadar24 misbehavior.
Please edit the file `opt/adsb/config/fr24.yml` so it reads like this:
```
version: '2.0'

services:
  fr24feed:
    image: ${FR24_CONTAINER}
    tty: true
    container_name: fr24feed
    restart: always
    ports:
      - ${_ADSBIM_STATE_FLIGHTRADAR_PORT:-8754}:8754
    environment:
      - BEASTHOST=ultrafeeder
      - FR24KEY=${FEEDER_FR24_SHARING_KEY}
      - MLAT=${FEEDER_FR24_MLAT}
    volumes:
      - /run/fr24feed:/run
```
Only the last two lines should be different...

dirkhh, 2023-10-16:
With the changes discussed here (and another one for dump978 that doesn't affect you), in the past hour on a moderately busy site, all containers combined have written about 4MB. So call it 100MB a day or 3G/month.
Dropping it further than this will lose the history / the ability to show tracks in the map.

andreagdipaolo, 2023-10-16:
That's way better!

[image]

[image]

I can confirm that now we have an average of 40-45 kB/s including my other containers, equal to around 20-25 kB/s for the airplanes. Maps, heatmap and tracks seem to work flawlessly.

dirkhh, 2023-10-16:
Well, I'm glad that it's trending in the right direction at least.
Thank you for alerting me to these suboptimal defaults, and helping me figure out what's going on.
If you run `iotop -a -o` and wait for a few minutes, are there any ADS-B related processes that stand out as creating a lot of writes? For me the only one left is `readsb` which is writing the history / trace data...

andreagdipaolo, 2023-10-16:
This is after stopping my other containers and leaving only airplanes:

[image]

[image]

After 1 1/2 hours this is the result:

```
    TID  PRIO  USER     DISK READ DISK WRITE>    COMMAND                                                                                                                                                                                                                
    318 be/3 root          0.00 B     23.45 M [jbd2/nvme0n1p1-8]
2868339 be/4 root          0.00 B      7.09 M readsb --net --quiet --lat my location
3887189 be/4 root          0.00 B      4.61 M fr24feed
2465372 be/4 root          0.00 B    184.00 K dockerd -H fd:// --containerd=/run/containerd/containerd.sock
 817349 be/4 root          0.00 B    124.00 K dockerd -H fd:// --containerd=/run/containerd/containerd.sock
 827743 be/4 root          0.00 B    124.00 K dockerd -H fd:// --containerd=/run/containerd/containerd.sock
    781 be/4 root          0.00 B    104.00 K dockerd -H fd:// --containerd=/run/containerd/containerd.sock
    805 be/4 root          0.00 B    104.00 K dockerd -H fd:// --containerd=/run/containerd/containerd.sock
2905502 be/4 root          0.00 B    100.00 K dockerd -H fd:// --containerd=/run/containerd/containerd.sock
    737 be/4 root          0.00 B     88.00 K dockerd -H fd:// --containerd=/run/containerd/containerd.sock
 823402 be/4 root          0.00 B     52.00 K dockerd -H fd:// --containerd=/run/containerd/containerd.sock
```

But if I check SMART after 15 minutes I still see 22MB written, 88MB/h, 2GB per day. More than you.

`Data Units Written:                 2,184,853 [1.11 TB]`

`Data Units Written:                 2,184,897 [1.11 TB]`

dirkhh, 2023-10-19:
I think we can close this with the release of v1.0.0

closed 2023-10-19 by dirkhh