dirkhh/adsb-feeder-image #67: [BUG] OVA Error - Line 16: Missing child element 'VirtualHardwareSection'.

bslatyer, 2023-10-24:
Hey @dirkhh,

Looks like there might be an issue/bug with the OVA in VMware vSphere 8.

My hosts are running VMware vSphere 8.0 Update 2 on the Intel NUC (13th Gen).

When I go to upload the OVA to the Host - it results in this error.

Error: Line 16: Missing child element 'VirtualHardwareSection'.

There was an error creating the import specification from the OVF file.

[image]

dirkhh, 2023-10-24:
The OVA is created under VirtualBox and there appears to be a slight incompatibility...

So when running the OVA under Fusion it will give you an error, but then when you click "retry" it completes.

I haven't checked if there's an equivalent workaround for esxi

dirkhh, 2023-10-24:
This article is a few years old, but it might give a path worth trying. I don't have access to an esxi box (I stopped working for VMware a while ago and when I left rebuilt my home lab, migrating from esxi to an all open source solution...), so I can't try this.
https://zhengwu.org/convert-virtual-machine-of-virtualbox-to-esxi/

bslatyer, 2023-10-24:
Hey Dirk,

No worries - thanks for that!

I'll give it a go later today and let you know the results.

bslatyer, 2023-10-24:
After removing the following SHA line from the .mf file
**SHA1 (adsb-feeder-x86-64-vm-v1.0.1.ovf) = 6004a2e27b678b312ee64bc6710862ee0ebd634b**

It just infinitely spins now
[image]

I'll try creating the VM in either Fusion or Workstation and then try uploading it directly to the ESXi cluster from the GUI.

bslatyer, 2023-11-06:
Still no luck with this one @dirkhh. Any suggestions?

closed 2023-11-06 by bslatyer