dirkhh/adsb-feeder-image #90: add option of feed from an external beasthost rather than USB connected SDR

kmm0, 2023-11-21:
As there are many existing users feeding flight data to single aggregators add the option of feed from an external beasthost rather than from a USB connected SDR.   Then adsb-feeder-image can be used on a separate host to feed additional sites without any potential to impact the existing feeder.

This is particularly useful for those wanting to run in a VM as typically the host with the SDR is mounted as near to the antenna as possible to reduce RF loss and this is typically not practical with hardware than can support VM.

dirkhh, 2023-11-21:
this is something I have considered in the past. definitely on my longer term plan.

dirkhh, 2023-11-23:
As I am trying to think this through a bit more... there are a number of scenarios that I could envision and I'm curious what you had in mind.

What seems fairly easy would be something like this:

Your existing feeder with the SDR attached opens `beast_out` on `port 30005`. If this is a docker-ized setup, make sure this is forwarded to the host. With the feeder image there is an option to do just that on the Advanced page.

In the feeder image on the VM you add an extra ultrafeeder arg (on the Expert page) that reads something like `adsb,feeder-with-SDR-IP-address,30005,beast_in`

I haven't tested this extensively, but from a quick attempt this should work just fine.
I can use one adsb.im Feeder Image to run the actual SDR, and another one to read the output from that, run the map, forward to aggregators, etc.

Is that what you had in mind?

Obviously this could be wrapped in more "beginner friendly" UI and stuff - but it does seem we can already do what you are asking about, no?

kmm0, 2023-11-23:
Your existing feeder with the SDR attached opens beast_out on port 30005. If this is a docker-ized setup, make sure this is forwarded to the host. With the feeder image there is an option to do just that on the Advanced page.

In the feeder image on the VM you add an extra ultrafeeder arg (on the Expert page) that reads something like adsb,feeder-with-SDR-IP-address,30005,beast_in

Yes, this is what I was thinking.

For the UI part, it's probably in the form of a toggle for SDR or external source with external source having input for beast host and beast port.

kmm0, 2023-11-23:
Adding the comment that external feed for UAT should be an option as well, toggle for SDR / external source with external source having input for UAT host and UAT port.

dirkhh, 2023-11-23:
Again - I politely decline to make this any more complex just to cater to UAT use. That's just such a marginal use case for a handful of people in the US. And allowing that to be on a remote device makes things significantly more complicated.

closed 2023-11-23 by dirkhh

kmm0, 2023-11-24:
confirmed external feeder is working, tested using adsb-im-raspberrypi64-v1.1.4-beta.2.img

commit 9da2d087 (2023-11-23): support two stage feeder backend without SDRs

This assumes that there is a second device that has the SDR(s) attached and
provides 'beast_out' on an open port.

Note that this does NOT attempt to also get UAT from a different device. This
is ONLY designed to allow you to get ADS-B / 1090 from a different system.

Fixes #90