dirkhh/adsb-feeder-image #91: add Raspberry Pi 5 support

kmm0, 2023-11-21:
add Raspberry Pi 5 support

dirkhh, 2023-11-21:
sure - once I have one to test with...

dirkhh, 2023-12-23:
I don't have a self-contained image, yet, but assuming you have a Pi3/4 around somewhere, you can do a neat two step setup.
- install the latest DietPi on the older RPi
- run the upgrade script that Micha is testing (details here: https://github.com/MichaIng/DietPi/issues/6676 but essentially you run `bash <(curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.meta/dietpi-rpi-firmware-migration')` once installation on the older Pi completes)
- shut down your old pi, plug the SD card into the Pi5
- now use `dietpi-software` to install the ads-b feeder up (`#141`)

[image]

Remember that when you install this as an app (instead of an image), the web UI is on port 1099, not port 80.

dirkhh, 2024-01-30:
It took a lot longer than expected - but a first test build is available as part of the v1.1.9-beta.2

closed 2024-02-29 by dirkhh