dirkhh/adsb-feeder-image #92: Add UAT support

kmm0, 2023-11-23:
Add support for UAT.

Either as an external source (should be easy) or 2nd SDR (harder as the SDRs need to be serialized).

Note, UAT is only used in the US and not all of the aggregators support UAT ingest.

dirkhh, 2023-11-23:
That is already supported.
Simply add a second SDR. As long as they don't have the same serial number, you'll be able to use one for UAT and one for ADS-B.
And if they have serial numbers 978 and 1090, they'll automatically be assigned the right role.

kmm0, 2023-11-23:
That's great, after looking again I did see mention of it on the using page of the site.

Might want to add mention that UAT is supported on the main page.

Once UAT is added, does the dashboard indicate the UAT is enabled for the aggregators that support it?

dirkhh, 2023-11-23:
The typical beginner (which is the target audience) would likely only be confused by mentioning UAT. It's US only, and in most cases even in the US only adds marginal additional reception.
There is no separate indicator if an aggregator accepts UAT traffic and whether they receive it.

closed 2023-11-23 by dirkhh