dirkhh/adsb-feeder-image #93: service "ultrafeeder" has neither an image nor a build context specified

nebriv, 2023-12-19:
Hi There,

Trying to use this on a Raspberry Pi 4 B. Fresh install straight from the image to SD card. 

I'm getting this error in /var/log/adsb-setup.log
```
adsb-system-restart: trigger docker compose up
calling docker compose for the enabled containers
time="2023-12-19T03:14:18Z" level=warning msg="The \"ADSB_SDR_PPM\" variable is not set. Defaulting to a blank string."
service "ultrafeeder" has neither an image nor a build context specified: invalid compose project
192.168.1.46 - - [19/Dec/2023 03:14:18] "POST /restart HTTP/1.1" 200 -
```

I get more or less the same error when running:
sudo bash /opt/adsb/docker-compose-start

And it does not appear as though any of the docker containers are up:
```
root@adsb-feeder:/var/log# docker ps
CONTAINER ID   IMAGE     COMMAND   CREATED   STATUS    PORTS     NAMES
```

dirkhh, 2023-12-19:
Which version are you trying to use.
Which exact image (please include the filename - there are three different images that would work on an RPi).
Does the network on your RPi work? Is it behind some kind of firewall, reverse proxy, or some other odd network environment?

nebriv, 2023-12-19:
I grabbed adsb-im-raspberrypi64-v1.1.4.img and also tried adsb-im-raspberrypi64-v1.1.5-beta.2.img. Same result

No odd network configs, the network on the pi (wired) does work, I'm able to SSH in remotely (uploading key via the web ui).

dirkhh, 2023-12-19:
It sounds like your Pi can't get to the GitHub docker repo. Or the repo is unwell (somewhat unlikely, but not impossible).

I'm not at my computer, so I can't verify right now.

nebriv, 2023-12-19:
Gotcha, I'll see if I can manually hit the repo to troubleshoot. Is it the ultra feeder repo, or this one?

dirkhh, 2023-12-19:
It's the ghcr.io repo that the docket images are pulled from

nebriv, 2023-12-19:
Well, it's working now! Glitch in the matrix I suppose. Maybe got caught by this:
https://www.githubstatus.com/incidents/476y2pqmb3hc

closed 2023-12-19 by nebriv