dirkhh/adsb-feeder-image #94: SDR support

johnboy00, 2023-12-23:
Regarding the list of supported hardware on the ADSB.im website, there's no such thing as a "_FlightRadar_ pro stick."  I'm sure you must be referring to the "**FlightAware** Pro Stick."  Is the FlightAware Pro Stick Plus also supported?

Would you consider adding support for the RadarBox "red stick" (UAT 978 filter + LNA)?

dirkhh, 2023-12-23:
All these interchangeable names... I think I conflated the RadarBox FlightStick and the FlightAware ProStick into the FlightRadar ProStick? I don't even know. 

But basically all of these are already supported, including the RadarBox red 978, the RadarBox green 1090, all of the FlightAware sticks, the ADSBExchange / Icarus stick... all of them.

dirkhh, 2023-12-23:
Updated and expanded.

dirkhh, 2023-12-23:
Oh, and thanks for the feedback, Should have said that FIRST. I really appreciate it. That's how errors get caught :)

closed 2023-12-23 by dirkhh