dirkhh/adsb-feeder-image #95: [BUG] RadarBox - Your machine doesn't have a serial number in /proc/cpuinfo

bslatyer, 2023-12-29:
Hey Dirk,

Me again :)

The RadarBox script located at /opt/adsb/rb-hack-setup.sh needs to be fixed when requesting a new share code or using an existing share code.

I'm using an x86_64 VM in VMware Workstation, which has previously been working.

Running the script manually doesn't work either.

This is the error I get below on any VM instance.

```
2023-12-29T23:11:24.115729000Z [2023-12-30 10:11:24.115][02-rbfeeder] 
2023-12-29T23:11:24.117614000Z [2023-12-30 10:11:24.117][02-rbfeeder] [ERROR] Your machine doesn't have a serial number in /proc/cpuinfo
2023-12-29T23:11:24.119177000Z [2023-12-30 10:11:24.119][02-rbfeeder]         You must resolve this for the container to run. Instruction to fix it can be found here:
2023-12-29T23:11:24.120844000Z [2023-12-30 10:11:24.120][02-rbfeeder]         https://github.com/sdr-enthusiasts/docker-radarbox/blob/main/version_0.4.3_workarounds.md#workaround-for-cpu-serial
```

[rbfeeder-2023-12-29T23-12-46.log](https://github.com/dirkhh/adsb-feeder-image/files/13796483/rbfeeder-2023-12-29T23-12-46.log)

Thanks,

Braeden

dirkhh, 2023-12-29:
Which version of the adsb.im feeder image are you using? These workaround should have been in place for quite a while.

bslatyer, 2023-12-30:
I'm running v1.1.5 (Stable)

dirkhh, 2023-12-30:
I know that this is always the least favorite answer... but I just installed an x86 VM from scratch, using the vmdk from the release. I waited for it to boot, configured it, and then had it request a key from RadarBox.
All this worked flawlessly.

So there must be something different about your setup. Given that you appear to have looked at the hack script already - do you have an idea what's going wrong in your case? Because without more information, I'm not sure how to help you.

bslatyer, 2023-12-30:
Ah, yes, the go-to line that QA hate: "It works on my machine." 😂

When the hack script is run on the VM - I get this output.

```
root@adsb-feeder:/opt/adsb# ./rb-hack-setup.sh 
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ss ht syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon rep_good nopl xtopology tsc_reliable nonstop_tsc cpuid tsc_known_freq pni pclmulqdq vmx ssse3 fma cx16 sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand hypervisor lahf_lm abm 3dnowprefetch ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdseed adx smap clflushopt clwb sha_ni xsaveopt xsavec xgetbv1 xsaves avx_vnni arat umip pku ospke gfni vaes vpclmulqdq rdpid movdiri movdir64b fsrm md_clear serialize flush_l1d arch_capabilities
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ss ht syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon rep_good nopl xtopology tsc_reliable nonstop_tsc cpuid tsc_known_freq pni pclmulqdq vmx ssse3 fma cx16 sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand hypervisor lahf_lm abm 3dnowprefetch ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdseed adx smap clflushopt clwb sha_ni xsaveopt xsavec xgetbv1 xsaves avx_vnni arat umip pku ospke gfni vaes vpclmulqdq rdpid movdiri movdir64b fsrm md_clear serialize flush_l1d arch_capabilities
```

The script is also adding the FEEDER_RB_THERMAL_HACK entry.

Which results in this one-liner getting added to the .env

`FEEDER_RB_THERMAL_HACK=/sys/class/thermal`

However, the script is failing to add the one-liner entry for `FEEDER_RB_CPUINFO_HACK` but seems to create an empty directory from the script called `/opt/adsb/rb/cpuinfo`

Thoughts?

bslatyer, 2023-12-30:
I should also add that running this line from the script results in this output

```
root@adsb-feeder:/opt/adsb# echo -e "Serial\t\t: $(echo $RANDOM$RANDOM $RANDOM$RANDOM | awk '{printf "%08x%08x", $1, $2 }')" >> /opt/adsb/rb/cpuinfo
-bash: /opt/adsb/rb/cpuinfo: Is a directory
```

dirkhh, 2023-12-30:
that's fascinating. 
I think I have an idea how this could happen in theory, but there is a bit of an order of execution problem here, i.e. this shouldn't be able to happen the way things are supposed to be called.
And since I don't know how this can possibly have happened, it's a little hard to figure out a way to work around it / prevent it from happening.
I brute-forced something for v1.1.6-beta.2 to see if maybe that does the trick for you.

dirkhh, 2023-12-30:
This bothers me...
> ```
> root@adsb-feeder:/opt/adsb# ./rb-hack-setup.sh
> flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ss ht syscall nx pdpe1gb [...]
...
> 
> However, the script is failing to add the one-liner entry for `FEEDER_RB_CPUINFO_HACK` but seems to create an empty directory from the script called `/opt/adsb/rb/cpuinfo`
> 
> Thoughts?

(a) nothing in the v1.1.5 script would create that directory
(b) the script had a bug where it didn't hide the result of the grep, but that result should only have been shown ONCE, not TWICE, as in your output.

This makes me wonder about the script that you have in that spot in the first place. It doesn't sound like it could reasonably be the script that ships with v1.1.5

Anyway, v1.1.6-beta.2 should work around any artifacts left over on your system and remove the `cpuinfo` directory that somehow got created for you.

bslatyer, 2023-12-31:
Here's the copy of the script from the affected machine running 1.1.5 (GitHub doesn't like .sh uploads)
[rb-hack-setup.txt](https://github.com/dirkhh/adsb-feeder-image/files/13800431/rb-hack-setup.txt)

I'll give the beta a try over the next day or two and let you know how it goes.

bslatyer, 2023-12-31:
Quick update - it's still the same error running on 1.1.6 B3

`[2023-12-31 06:44:12.696][02-rbfeeder] [ERROR] Your machine doesn't have a serial number in /proc/cpuinfo`

[temp_container-2023-12-31T06-44-39.log.gz](https://github.com/dirkhh/adsb-feeder-image/files/13800441/temp_container-2023-12-31T06-44-39.log.gz)

dirkhh, 2023-12-31:
Hi Braeden. I can see from the output that the hack isn't applied to the temp_container. What I can't understand from the code that is running is "How the heck is that possible".
I can't reproduce it here and I'm kinda running out of ideas. 
I once again started from a clean v1.1.6-beta.3.
I go to the aggregator page and request a RadarBox key.
In a shell I verify that `/opt/adsb/rb` is created and that the `cpuinfo` file is created and that it contains a serial at the end.
The request completes and I have my fifth new RB key since I tried to fix this issue. RB will soon be mad at me :)

I have absolutely no idea why this works when I start from scratch and why it doesn't work for you.
Clearly something is different for you. But heck if I know what.

commit fbdd8a61 (2023-12-30): attempt to fix strange RB issue on x86

This makes no sense. This should never be a directory, but a user reports that
it is. So instead of arguing why it can't be, let's just make sure that it gets
replaced by the fake cpuinfo file if and when that happens.

Fixes: #95
closed 2024-01-02 by dirkhh