dirkhh/adsb-feeder-image #103: Feature request: select or slider input for gain

Ottergoose, 2024-01-26:
Per autogain docs, acceptable gain values are: 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 and -10 for auto. It'd be slick if these allowable values were pre-populated in a select or slider input. Thanks!

dirkhh, 2024-01-26:
Hey Nick, thanks for the sponsorship!

I talked to Matthias (readsb author) about this, and while these are indeed the correct value, they are also pointless as the app simply picks the closest one... and you do NOT want to use -10 for ADS-B, so that isn't offered.

So while this may seem useful, it really doesn't make any difference. Pick a number between 0 and 50 if you really can't make autogain work for you - but anything beyond that is just going to confuse people... because the next person will say "well, 20.7 is a bit too low and 22.9 is definitely too high, can you implement 21.8?"

Ottergoose, 2024-01-26:
You're welcome!

Good to know about -10.

That said... my counterpoint would be why not offer valid values for ADS-B to experts so they don't have to cross reference notes in a different Github repo when they're playing around to fine tune things? Furthermore, it provides an explicit option to get back to "autogain" without forcing user to type it in? I understand the potential support hassle, but would imagine you reduce support headaches by providing and explicit list of allowable values - it'd certainly make tinkering easier for me.

```
<select id="gain">
<option value="autogain">Automatic</option>
<option value="autogain">0.0</option>
<option value="autogain">0.9 </option>
<option value="autogain">1.4 </option>
<option value="autogain">2.7 </option>
<option value="autogain">3.7 </option>
<option value="autogain">7.7 </option>
<option value="autogain">8.7 </option>
<option value="autogain">12.5 </option>
<option value="autogain">14.4 </option>
<option value="autogain">15.7 </option>
<option value="autogain">16.6 </option>
<option value="autogain">19.7 </option>
<option value="autogain">20.7 </option>
<option value="autogain">22.9 </option>
<option value="autogain">25.4 </option>
<option value="autogain">28.0 </option>
<option value="autogain">29.7 </option>
<option value="autogain">32.8 </option>
<option value="autogain">33.8 </option>
<option value="autogain">36.4</option>
<option value="autogain">37.2</option>
<option value="autogain">38.6</option>
<option value="autogain">40.2</option>
<option value="autogain">42.1</option>
<option value="autogain">43.4</option>
<option value="autogain">43.9</option>
<option value="autogain">44.5</option>
<option value="autogain">48.0</option>
<option value="autogain">49.6 </option>
</select>
```

Just my $0.02, appreciate your efforts as always! I hope the provided HTML will help entice you further. 😂

dirkhh, 2024-01-26:
I see the value of a dropdown box. Hmm. 
I do think that this might potentially confuse the beginner, but I'll play with it.

(and I repeat, no one has to cross reference the exact values... type in `40` and you get `40.2`... type in `39` and you get `38.6`)

dirkhh, 2024-02-05:
I thought about this some more. This would require that I split out airspy and RTL based SDRs. And it will create a fairly inconsistent user experience. It's unclear (I find contradicting data) how this would work for UAT, potentially creating a third different experience. I don't think that is an improvement.

Ottergoose, 2024-02-12:
That's entirely reasonable; cheers!

closed 2024-02-12 by Ottergoose