dirkhh/adsb-feeder-image #104: What ports need forwarded for BEAST/MLAT (or are they even open on the feeder image?)

ilikenwf, 2024-01-31:
In my firewall logs I notice some of the aggregators hitting ports like 30004, 30001, 31090 - do I need to forward these to my feeder for MLAT/BEAST purposes?

If there is a case like this or any others it would be nice to have it documented somewhere here and in the feeder image web console, perhaps?

ilikenwf, 2024-01-31:
Some of the input ports listed here...should we be listening on those to outside addresses? Otherwise why would some of the adsb sites be trying to connect to my feeder?

https://github.com/dirkhh/adsb-feeder-image/blob/2e70efaf4640b0a5835791b37d8c71ec291f85bb/src/modules/adsb-feeder/filesystem/root/opt/adsb/docker-compose.yml#L12C9-L12C14

dirkhh, 2024-01-31:
No external ports should need to be opened. All my feeders run behind completely locked down (inbound, that is) firewalls.

ilikenwf, 2024-01-31:
Good to know, same situation here...makes it weird, though, that some of the ADSB mapping sites are trying to connect to my MLAT and BEAST ports then, no?

dirkhh, 2024-01-31:
I'm curious who's connecting. I assume you have done RDNS lookups?

ilikenwf, 2024-01-31:
Yes - 

radarplane.com
another one hosted on hetzner with no domain 116.202.107.112
adsb.lol (a subdomain k2.adsb.lol)
another one with no real domain 78.46.234.18
one on AWS 34.222.122.143
132.145.112.6

ilikenwf, 2024-01-31:
Makes me think that we may need to optionally open ports for further MLAT/BEAST operation? May have to see what other feeder images for specific sites are doing?

dirkhh, 2024-01-31:
@katlol - I'm curious about the connections from your server back to a feeder... how is this supposed to work? Almost any feeder I can think of sits behind a typical firewall that would block direct access. I must be missing something...

ilikenwf, 2024-01-31:
You know what, I may be stupid...

I just noticed that the log says this was on my LAN side...and it's the IDS/IPS. So I think it may be, in fact, that suricata just falsely flagged these.

dirkhh, 2024-01-31:
That would make more sense - I trust @katlol not to do something odd like this - but I'm traveling right now and can't easily access my own firewall logs to see if I might also have such connection attempts, so I took your word for it... 🤷🏼‍♂️

Sorry for the noise, Katia.

ilikenwf, 2024-01-31:
Sorry for the mistake - I'm in between caffination sessions.

katlol, 2024-01-31:
For beast	, it's a (mostly?) one way communication from you to the aggregators via beast, for MLAT, you get mlat results back on the same port you connected for MLAT. 

Either way, you are not expected to port forward any ports nor does adsb.lol connect to your IP

dirkhh, 2024-01-31:
Thanks for confirming, Katia.
That's what I had assumed.

closed 2024-01-31 by dirkhh