dirkhh/adsb-feeder-image #109: Possible to use this image to grab data from FlightFeeder and share?

cartpauj, 2024-02-27:
I have a FlightFeeder from FlightAware. I've setup a Pi that listens on the beast port of the feeder and shares the data to FR24 and adsb.fi

I'm interested in using your image to help simplify who I share with.

Is it possible to use the FlightFeeder beast port as the input for this?

dirkhh, 2024-02-27:
I have no experience at all with the flight feeder device. I assume this would be possible with some tweaking, but it is not something that is currently supported.

cartpauj, 2024-02-27:
The flight feeders use dump1090, so the beast data is coming out of dump1090 on port 30005

dirkhh, 2024-02-27:
So for a technically minded person that's likely something that they could add without much hassle.
Adding it as a general options for everyone to me seems like... a very small audience, really.

closed 2024-03-09 by dirkhh