dirkhh/adsb-feeder-image #111: Back Up feature crashes the RPi3b (1.1.9 stable)

ramphex, 2024-02-28:
Currently on 1.1.9 stable, wanted to back up before upgrading to 1.2.0 stable. Every time I go into the Back Up section and press "Back Up", the browser loading bar starts going, and then hangs, after some time causing the RPi3b to crash (only red light on, no web access, doesn't show up on router client list anymore)

dirkhh, 2024-02-28:
I did a number of backups today and am confident that the code itself is correct.
But depending on how long you've been using the feeder, the backup of the statistics is a rather significant amount of data. I'm wondering if somehow the process of creating the backup file causes the Pi to run out of memory and not deal gracefully with the problem. I do have a number of 0.5G and 1G devices - I'll try to simulate the problem here to see if that is the issue (and how to work around it).

Thanks for the report!

ramphex, 2024-02-28:
Any chance to split it into Back Up (Full) and Back Up (Settings Only)?

dirkhh, 2024-02-28:
That would certainly easy to do.

dirkhh, 2024-03-02:
I still haven't been able to reproduce a crash, even on a 512M Pi3a -- but the "config only" backup is now part of the v1.2.1 beta series and will land on v1.2.1 (hopefully this weekend).

commit 19a1d0e2 (2024-02-29): offer full and config-only backup

On a long running feeder the backup can be quite huge, and apparently
on low memory systems that can cause crashes. So allow a config-only
backup as well.

Fixes #111
closed 2024-03-02 by dirkhh