dirkhh/adsb-feeder-image #112: which image is for RockPi?

mattfox27, 2024-03-03:
which image is for RockPi?

dirkhh, 2024-03-03:
I haven't had a download for the RockPi in pretty much forever, so I stopped making those images.
If you tell me which model you have, I'll be happy to build one for you

dirkhh, 2024-03-09:
let me know if you want help / an image for a specific board. I'll close the issue for now since I haven't heard back.

closed 2024-03-09 by dirkhh

mattfox27, 2024-03-09:
Its OK I just switched to a Raspi Pi 4

dirkhh, 2024-03-09:
well that of course is extremely well supported.
When I started working on this image about a year ago, RPi were almost impossible to get - unless you paid crazy scalper prices. So I tried to support several other boards. And interestingly some of them are "mildly popular", and others never really got any users - so I dropped making those. I would have happily added one for you, but this is of course easier :)

mattfox27, 2024-03-09:
I appreciate that, I was just having all sorts of issues with that rockpi, so I ended up just buying an rpi4 off Amazon and installed and it's working great.