dirkhh/adsb-feeder-image #116: SDR Serial # detected as bNumConfigurations

wmnl25, 2024-03-06:
Hi, 

I have an issue on my armbian box where the SDR serial is detected as "bNumConfigurations". This causes ultrafeed to fail. 

[image]

`[2024-03-06 15:15:04.383][readsb] rtlsdr: no device matching 'bNumConfigurations      1' found.`

When manually setting the serial to "0" in config/docker-compose.yml (- READSB_RTLSDR_DEVICE=0), ultrafeeder starts to work.

Running [ADS-B Feeder Image](https://adsb.im/home) v1.2.1(stable) with containers from 2024-03-06T14:54:18+01:00, running on Rockchip RK3318 BOX.

This is no issue on my other box: "Running [ADS-B Feeder Image](https://adsb.im/home) v1.2.0(stable) with containers from 2024-03-03T09:44:21+01:00, running on Raspberry Pi 5 Model B Rev 1.0."

Both installed through the install script, not using a pre built image.

dirkhh, 2024-03-07:
interesting... I thought that all dongles came from the factory with a default serial number.
Now of course I would have LOVED to see the `lsusb` output before you set the serial number in order to be able to detect that case and be confident that I get it right.

But yes, setting a serial number is an obvious workaround.

dirkhh, 2024-03-09:
So I have tried to implement something that **should** address this - but of course I can't test it.
But I'll close this for now.

closed 2024-03-09 by dirkhh