dirkhh/adsb-feeder-image #118: Feature request: Add optional offset for autogain

ilikenwf, 2024-03-08:
I have both a UAT and standard dongle, both have inline filters and amps, but I added a SAWBird into the mix. Because the sawbird adds I think 32dB or so, it would be nice to be able to negative offset the determined autogain value by about that much.

Right now I'm having to manually set it...

I'm not even sure if the SAWBird makes a huge difference or any at all yet, but I'd love to be able to use autogain again.

dirkhh, 2024-03-08:
Simply reset autogain and it should figure out a good range.
Unless you reset it, it will only adjust VERY slowly, but once you reset it, it will try to quickly find the approximate range and then improve over time.

ilikenwf, 2024-03-08:
Other feeders using autogain also have noticed the issue, but manual has to be used with the sawbird, autogain always ends up picking values right now that makes things inoperable otherwise.

dirkhh, 2024-03-08:
Well. Umm. I have a sawbird+ on two of my feeders, one with a blue stick with LNA, the other with an airspy.
In both cases I'm running with autogain without a problem.
I am aware of one user who complains about the results of autogain and insists on setting manual values, but the data they posted on discord didn't seem to indicate to either me or wiedehopf that they were getting a significant benefit from that.

If in your case autogain (after reset) turns the setup inoperable I'd be curious about a full log (from autogain reset until autogain settles). I can envision a scenario where you are very close to strong signals and where with double LNA you might already massively over-powering the SDR at the default starting value.

But that would be a scenario that I would describe as "doctor, it hurts when I poke a spoon into my eye". In other words - why would someone want to add a sawbird in such a scenario.

ilikenwf, 2024-03-12:
Where are the autogain logs kept? I've not poked around much inside the docker containers.

dirkhh, 2024-03-12:
From the UI you can go to Logs (top right corner) and there click on Ultrafeeder. That will show you what it's doing with autogain (and a lot of other things).

wiedehopf, 2024-06-16:
How is the autogain behaving now? Can you give it another try i believe there were some major changes in the meantime.

ilikenwf, 2024-06-17:
So I had 19.4 set as my gain previously. I tried resetting and setting autogain, and got very low message rates, like 10.1/sec.

Went back to 19.4, then saved, then back to autogain and it seems to work - so it appears to be starting the calculations based on my initial 19.4 figure, perhaps? Seeing good results doing this...

wiedehopf, 2024-06-18:
You can check the 24h misc graph to see what's going on.

ilikenwf, 2024-06-18:
I'll take a look tomorrow if I remember during higher traffic times to be sure my message rate is high, I'm over 500 during busy hours...am kinda rural but see something like 200-250 miles away.

[image]

[image]

Going by this do I need to use my old gain level?

[image]

wiedehopf, 2024-06-18:
Nope, not sure why you had gain that high before, this looks more reasonable.

Resetting autogain will make it start high but it should go down pretty quickly.
Feel free to give it a try.

978 is a different story, that might need to stay on manual.

closed 2024-06-24 by wiedehopf

ilikenwf, 2025-01-12:
@wiedehopf Sorry to bring up a dead topic; should autogain be solid for both with the sawbird? I also question if I should even be using a sawbird since I was seeing close to similar range without one way back when - could do a/b testing sometime...

That said after a few power outages for whatever reason my feeder image system decided to forget settings and I had to restore from backup. Currently autogain is in use on both UAT And 1090...but I wanted to check if the situation on those is solid and I can just autogain both now or not.