dirkhh/adsb-feeder-image #123: FlightAware hangs after TLS error; dashboard status remains normal

Ottergoose, 2024-03-15:
Looks like a TLS issue caused my connection to FlightAware to hang. Viewing my FlightAware dashboard on their website, it says no data has been sent since the timestamp here in the logs:

```
[piaware] 2024/03/15 10:35:24 15409 msgs recv'd from socat (152 in last 5m); 15402 msgs sent to FlightAware
[piaware] 2024/03/15 10:35:31 TLS alert (read): bad record mac
[piaware] 2024/03/15 10:35:31 TLS error: 0
[piaware] 2024/03/15 10:35:31 Lost connection to adept server at piaware.flightaware.com/1200: error reading "sock55a07775f0": software caused connection abort
```

The dashboard on the local feeder ADSBim homepage doesn't show any problem (green checkbox), and if I click on the feed link, it shows green across the board at :8082, until about 5 seconds later, when all of the status boxes switch to red, and the message on screen says "PiAware doesn't appear to be running!"

Overall CPU usage on my Pi 3 B+ goes from ~25% to ~50% and has remained there.

Not an urgent issue; if there's anything else you'd like copy/pasted here for debugging, please let me know. I'll hold off on rebooting until I've had a chance to help clarify anything as needed.

Appreciate it!

dirkhh, 2024-04-17:
is this still happening - we switched to a newer Piaware container, but I don't know if that would have affected this at all...

wiedehopf, 2024-06-11:
I've read on the FA forum, might just be some error FA can't find.
They've put a watchdog as a workaround ... not sure what exactly.
Possibly we can do something similar for the container.

wiedehopf, 2024-06-11:
https://github.com/flightaware/piaware/pull/87

So this is the relevant thing.
I'm not sure how to implement this in the piaware container to be honest.
Does piaware put out log messages in some interval?
I suppose the container could watch for that and do a reset of the process if it doesn't see  the log message.