dirkhh/adsb-feeder-image #125: Feeder containers not running after update to v1.3.0

Freekers, 2024-03-22:
After updating from v1.2.3 to [v1.3.0](https://github.com/dirkhh/adsb-feeder-image/releases/tag/v1.3.0), all feeder containers were not running anymore. The webinterface was reachable, but showed a red check-mark in front of all aggregators. The map was not loading (timeout) either and neither were the logs. `docker ps` showed 0 running containers. A reboot of the RPi did not solve the issue.

I noticed the 'Basic Setup' page had changed slightly, as an additional option was added for ' Which semi-anonymous aggregators do you want to feed?'. I fixed the issue by simply pressing the blue 'Submit' button again (without changing anything) on the Basic Setup page.

dirkhh, 2024-03-22:
That probably means that a setting didn't get transferred correctly.
I would have loved to see logs and your config.json, but I guess that's all gone by now.

Oh well, glad you got it working

ramphex, 2024-03-22:
Can confirm a similar issue after the upgrade. A extra reboot did start the feeders back up though.

Freekers, 2024-03-22:
> That probably means that a setting didn't get transferred correctly. I would have loved to see logs and your config.json, but I guess that's all gone by now.
> 
> Oh well, glad you got it working

For future reference, which logs would you be interested in? `/opt/adsb/adsb-setup.log`? Because that one indeed is gone already.

dirkhh, 2024-03-22:
Yes, what I would love to get access to is
- `/opt/adsb/adsb-setup.log`
- `/opt/adsb/config/.env`
- `/opt/adsb/config/config.json`

In an unrelated conversation @wiedehopf pointed out that there may be an issue where depending on exactly which containers you run and how, the order in which I currently update things may be problematic. So we may have a lead on what's going on. Dunno.

dirkhh, 2024-03-22:
And I just was able to reproduce it. @wiedehopf was correct in his gut feeling.
Issue identified, v1.3.1 should be coming fairly soon which should fix this.

Also - no reboots or other acrobatics necessary. All it would have taken to get things working is to click on the 'restart containers' button on the Expert page.

closed 2024-03-22 by dirkhh

Freekers, 2024-03-22:
Alright that's good to know, thanks. Next time I face an issue, I'll make sure to save the logs first before touching anything. Cheers.