dirkhh/adsb-feeder-image #129: how-to suggestion

wiedehopf, 2024-03-23:
```
https://adsb.im/howto

Download the latest release for the correct board from the [ADSB Feeder Image Release Page](https://github.com/dirkhh/adsb-feeder-image/releases/latest) In our example we will pick the adsb-feeder-lepotato-...img.xz file.
```
This link should possibly open in a new tab?

wiedehopf, 2024-03-23:
And possibly add this:
```
If you're using a Raspberry Pi 3 or 4, use one of these two images:
adsb-im-raspberrypi64-[....].img.xz
adsb-im-iob-raspberrypi-dietpi-64-[....].img.xz
```

wiedehopf, 2024-03-23:
And one further note, the How-to says how to configure the WiFi for Raspbery Pi OS but not for DietPi.
In that explanation, a link to the FAQ which explains how to do it for dietpi might not hurt (https://adsb.im/faq)

dirkhh, 2024-03-23:
I think I made all of the suggested changes.

closed 2024-03-23 by dirkhh