dirkhh/adsb-feeder-image #138: Housekeeping for header navigation and multiple setup forms

Ottergoose, 2024-03-28:
Even as a frequent user of the software, the number of setup / config links in the header navigation is confusing, and one of the functions I most frequently access (Stats) isn't listed in the navigation.

I might suggest the following:
- Basic Setup and Aggregators could be a single form / page, Settings
- Advanced and Expert could be a single form / page, Advanced Settings
- Remove the Restore link / form, as it's the first option in the current Basic Setup page
- Add a header link to Stats (:8080/graphs1090/)

dirkhh, 2024-03-28:
UI design is more an art than a science. And I have no claim to being any good at it.
I have, however, worked with some pretty awesome UI developers (and as a result even hold some patents on UI design - but that's a different story).

There is a theory in UI design that talks about the amount of "stuff" that you want to have on a page and how you need to try to make it connected in order for it to make sense to people. And how the first page of your UI should have as few items as are needed in order to complete a first logical step.

With that in mind
 - you can completely configure a feeder with just the first basic setup page and never go to the Aggregator page. And when you want to add or remove an aggregator, there is absolutely no point in being on a page where you set your location or time zone. So combining Basic Setup and Aggregators makes NO sense to me at all. It creates information overload and combines fundamentally unrelated concepts on one page.
 - Advanced and Expert will actually be split into THREE pages in the upcoming v2 - because they deal with three distinct kinds of things (visualization settings, hardware / SDR settings, and options that are mainly about the SBC) and those are different depending on whether you have an integrated feeder (what most people have), a micro feeder (just SDR+antenna - everything else happens elsewhere) or a stage 2 instance (that's the 'elsewhere' - which can map to multiple micro feeders). So no, I will definitely NOT be integrating Advanced and Expert
 - Removing backup and restore makes total sense to me I'll do that and put that on the expert page
 - the page that everyone goes to the most is very personal - I really didn't want the map link there but was convinced that this was the most important thing that people want to look at. I'd argue that the heatmap is far more likely to be the second most useful page and that once a feeder is running well no one (of the broader audience) will ever look at the stats. And the data geeks will have their own telegraph/prometheus/grafana/whatever instance, anyway, and won't look at the stats, either. I'm not saying 'no' to the stats link in the top level menu - but let's just say I'm not yet convinced...

Ottergoose, 2024-03-28:
Fair points! In that case, now that the software has grown so much, worth again considering a hierarchical menu in the nav and splitting settings up and labeling them unambiguously, perhaps something like this?

Flight Map (Tar1090 Map)
- Heatmap
- Timelapse Replay
- Tracks
- UAT Map

System Graphs (existing )
- Logs

Settings (existing Basic Setup)
- Radio Management (SDR stuff, Bias-T, SDR gain)
- Data Sharing (renamed "Aggregators"; move privacy tag there?)
- Map Options (HeyWhat's That, range rings, LOL Route API)
- System & Networking (reboot, software updates, root access, SSH)

dirkhh, 2024-04-04:
These are all really good ideas. Given the changes that I'm making right now, I'm trying to figure out how to integrate them in a seamless manner - as now is the time to do it. I don't have a final answer, yet - but wanted to let you know that I'm working on it.

Ottergoose, 2024-04-04:
Looking forward to seeing these, thank you @dirkhh!

dirkhh, 2024-04-04:
If you feel adventurous and have a spare system, there are test builds you could play with...

Ottergoose, 2024-04-12:
Sorry Dirk, was busy getting ready for an eclipse viewing out of town - I'm back on the grid now!

Love the new menu, thank you for making those changes! Additional suggestions:

- Rename "Main Map" to "Live Map" or "Flight Map"
- Rename "Graphs" to "Additional Maps"
- Rename "SDR" to "SDR / Radio Dongles" or "Radio Management (SDR)"; ("SDR" alone is not friendly to novices)
- Move "Map Options" to be nested under "Setup"
- Move "Data Sharing" to be nested under "Setup"
- Rename "Logs" to "System Data"
- Move "Stats" to be nested under "System Data"

dirkhh, 2024-04-12:
So in summary, you hate absolutely everything about it 🤣

I like "Live Map"
Most of your other suggestions are longer and that's not the direction I will go.
Also, if someone doesn't understand what an SDR is, then none of the docs, none of any of this will make sense to them.
Map Options and Data Sharing must stay top level

Ottergoose, 2024-04-12:
I don't hate any of it, it's a great improvement! 

The live map should be it's own top-level link, as I'm fairly confident everyone using the software frequently wants to see that map; however, "graphs" is a poor label for maps - they're additional maps, label them as such.

"SDR" is too concise for some novice and long-term users alike. It's still not the first thing I think of when I think of them, and I've been playing with this stuff for almost a decade now. You'll have to trust that there are some other users who see/approach this differently than you're intuiting here.

"Map Options" and "Data Sharing" are setup things - I see why you'd want them at the top level as they're fundamental functions for the software, but, as you said earlier, not everyone has the same things they want to access all of the time (beyond the flight map), so I would argue all of the setup/configuration stuff should be properly nested under "Setup." 

For example, I wouldn't imagine most users are accessing the Map Options page more than once or twice - the HeyWhat'sThat rings, route API, and resetting the range rings aren't things that require frequent access - having them as a top level link doesn't make sense in that context.

Again, this is what I propose, which is what you already built, just clarified and adjusted slightly:

- Live Map
- Additional Maps
  - Heatmap
  - Tracks
  - Replay
- Setup
  - Basic
  - Map Options
  - Data Sharing
  - Radio / Dongles (SDR)
  - Expert
- System
  - Logs (Dozzle)
  - Support Info
  - Share Diagnostics
  - Stats

I don't believe any part of that is too verbose, I think it's just a logical and broadly approachable topology of what the software has grown into now.

Again, appreciate your work, and this is all offered constructively; hope you'll consider chewing on it a bit before finalizing any decisions. Thank you!

dirkhh, 2024-04-12:
In stage 2 mode, Data Sharing and Map Options become drop downs with an individual page per micro feeders.
Moving them under Setup makes the whole thing completely unusable on a device.
I'll look into the 'System' label and moving stats there.

closed 2024-05-16 by Ottergoose