dirkhh/adsb-feeder-image #157: Missing mobile nav icon after upgrading to V2 beta from 1.3.6

Ottergoose, 2024-04-12:
The mobile/narrow menu .navbar-toggler button is present (I can see an element when I inspect the page, and I can click/tap on it), but the icon isn't visible. I can't see it on my desktop or mobile device.

dirkhh, 2024-04-12:
that's another one of the font-awesome icons. the styling framework that I used simply assumes those are there.
Need to find a replacement for that.

Ottergoose, 2024-04-12:
From viewing the source, I see there's an SVG defined locally, `--mdb-navbar-toggler-icon-bg: url("data:image/svg+xml;charset=utf-8,%3Csvg xmlns='http://www.w3.org/2000/svg' viewBox='0 0 30 30'%3E%3Cpath stroke='rgba(0, 0, 0, 0.55)' stroke-linecap='round' stroke-miterlimit='10' stroke-width='2' d='M4 7h22M4 15h22M4 23h22'/%3E%3C/svg%3E");`, so would think the fallback would/could/should work, but you know those frameworks better than I do now.

Ottergoose, 2024-04-16:
This has been resolved in one of the more recent builds, thank you @dirkhh!

closed 2024-04-16 by Ottergoose

dirkhh, 2024-04-16:
But now it can't figure out that 10 is bigger than 9🤦🏻‍♂️