dirkhh/adsb-feeder-image #160: Low disk space / Purge logs, installers, etc

ramphex, 2024-04-14:
After updating to 2.0.0.b4 I have noticed a new message on the homepage:

> You are running low on disk space on the** your feeder. This can lead to odd problems and even crashes. Consider upgrading to a larger storage device.

Is there a way to implement a feature to purge old logs and installers, or whatever occupies the largest amount of space? This setup is running on an 8GB card. 

** grammatical error

wiedehopf, 2024-04-14:
Try a beta update now, it would seem the docker system prune didn't remove the tagged images that were not being used.

ramphex, 2024-04-14:
[image]

wiedehopf, 2024-04-14:
Yeah that might be an issue.
You can run this manually and it should clear up space:
```
sudo docker system prune -a -f
```

ramphex, 2024-04-14:
> Total reclaimed space: 3.071GB
> root@adsb-feeder:~# 

Down to 58% used, updating to beta 6 now.

closed 2024-04-14 by wiedehopf