dirkhh/adsb-feeder-image #166: Roboto font still referenced

afoeder, 2024-04-18:
In 32916f7 the remote call to Google has been removed. However, the `Roboto` font is still referenced as font-family in [/src/modules/adsb-feeder/filesystem/root/opt/adsb/adsb-setup/static/css/mdb.min.css](https://github.com/dirkhh/adsb-feeder-image/blob/c55f0d314dfe7ed2d4700e84ce2ac6a0ad1f72c2/src/modules/adsb-feeder/filesystem/root/opt/adsb/adsb-setup/static/css/mdb.min.css) (search for `--mdb-font-roboto`).

I'm not familiar with the mdbootstrap.com ecosystem, if you know the cleanest way to remove it I'm open for suggestions :)

dirkhh, 2024-04-18:
So - the neat thing about how these fonts work is that the browser will substitute the font with something "appropriate" for your platform on the fly. MDB themselves say that the Roboto font is optional and have tried to set things up to ensure a reasonable alternative is picked.
(the same is not true for FontAwesome, unfortunately)

That said, I'm not opposed to to #167 - it just requires that we do the right thing from a license perspective.

afoeder, 2024-04-18:
fully agree and I won't be mad if we didn't pick up #167 — in any case I'm able to familiarize with the project :)

My inner "Monk" just doesn't fee well when the CSS asks for Roboto and it's inherently not there. The CSS fallback is rather meant for network outages etc not not break the whole site suddenly. But license compliance is 100% necessary, I agree!

dirkhh, 2024-04-18:
🙃
You are of course correct that this isn't ideal.
I'm not sure which direction I prefer, TBH.
I might be just very slightly biased towards keeping it simple and not adding more stuff - but then I should just look into removing Roboto, period.

PR #167 (2024-04-26): Reintroduce Roboto statically
This reintroduces the Roboto font
as local files after the hyperlink
to Google has been removed in 32916f7.

Fixes #166.
commit 13526dd5 (2024-04-26): Reintroduce Roboto statically

This reintroduces the Roboto font
as local files after the hyperlink
to Google has been removed in 32916f7.

Fixes #166.
closed 2024-05-04 by dirkhh