dirkhh/adsb-feeder-image #169: Unable to run two SDRs, multiple SDRs with same serial number error

ramphex, 2024-04-24:
On 2.0.0 stable, I plugged in a second SDR to run it for 978 but it will not let me save the settings in the SDR section. It just unchecks the second SDR every time (after refreshing the page) after I check it to be 978 and press "I'm done here - take me to the feeder homepage".

There's also the following message

> There are multiple SDRs with serial number 1090. This will not work correctly. Please ensure that all SDRs have distinct serial numbers.

I tried to change the serial of the second SDR to 978 with rtl_eeprom but it comes back with the following

> usb_claim_interface error -6
Failed to open rtlsdr device #0.

lsusb output:

> Bus 001 Device 022: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 021: ID 0bda:2832 Realtek Semiconductor Corp. RTL2832U DVB-T

rtl_test output:

> Found 2 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 1090
  1:  Realtek, RTL2832U, SN: 1090
Using device 0: Generic RTL2832U OEM
usb_claim_interface error -6
Failed to open rtlsdr device #0.

I want to use 2838 as 978 and 2832 as 1090.

ramphex, 2024-04-24:
Alright, I was able to change its serial utilizing another machine, then plugged it back into the Pi, and now no more serial number conflicts. I'm not sure if some container was trying to use it or what the issue was when trying to change the serial number via the Pi. I can't imagine many people running two dongles, but maybe have some sort of setting to be able to change serial numbers from the expert page or a link to a guide for more info when there's two with the same serial number (instead of just a warning).

dirkhh, 2024-04-24:
I looked into that (and yes - it is certainly a somewhat rare issue). The problem is that you need to be able to "own" the SDR in order to make that change - and there are a couple of interesting corner cases that make this a bit harder than it should be. Your report reminds me to look into this some more. Thanks.

dirkhh, 2024-11-07:
To comment here as well. I looked into how to do that half a year ago when this issue was raised.
And especially since a fair number of people use the feeder image not as an image but as an app on DietPi and other distros, the corner cases are really frustrating and hard to get right.
Yes, in the US the usecase of two SDRs for ADS-B is somewhat common (actually, not very common, but at least reasonable). But any usecase where people have more than that is clearly out of scope.

dirkhh, 2025-06-19:
... and, v3.0.0 will indeed support this corner case situation. So much for "clearly out of scope", I guess 🤣

closed 2025-07-04 by dirkhh