dirkhh/adsb-feeder-image #177: Restore more than just the config

ramphex, 2024-04-28:
On 2.0.0, on the Back Up page, I have options of Config, Config + Graphs, Full Backup. When I try to restore though, it asks me to select a file, assuming it'll only accept the config file. How do I go about restoring config + graphs or a full backup?

wiedehopf, 2024-04-28:
You upload the zipfile, then check all the boxes?

ramphex, 2024-04-28:
Hmm. So on MacOS Safari downloads the backup as a folder (assuming it automatically extracts the zip into a folder). I just tried it on Chrome and it downloaded the backup as a .zip file.

If anyone else runs into this issue, go to Safari Settings and uncheck 'Open "safe" files after downloading' in the General section.

wiedehopf, 2024-04-28:
https://github.com/awesomemotive/easy-digital-downloads/issues/1814

One option would be to add a note like someone did in the thread above:
> Note for Safari users: If clicking a download link saves a folder to your hard drive instead of a zip file, try Control-click or right-click and select "Download Linked File". Only zipped plugins are accepted in WordPress.

The image doesn't need the file to be named .zip
Thus another option would be to name it .backup for safari.

What a terrible automation.

dirkhh, 2024-04-28:
I never understood why someone would voluntarily use Safari. It's a terrible browser.
But yeah, let's just change the suffix of the file we return / expect to be `.backup`

wiedehopf, 2024-04-28:
Is there an easy way to detect it's Safari and only then change the suffix?
The restore would just accept both suffixes.

wiedehopf, 2024-04-28:
Of course safari is in the chrome user agent string ... yeah not sure browser detection is worthwhile.
Apparently the detection logic is: safari + chrome in the string -> chrome ... safari in the string but not chrome --> safari.

dirkhh, 2024-04-28:
Additionally, browsers and other user agents lie all the time. 
And it seems broken to send different files names, depending on browser. 
What's wrong with changing the name in general?

wiedehopf, 2024-04-28:
Yeah let's just change it in general then.
I liked the recognizable zip filetype just for clarity what it is.
But not worth the pain.

Not sure if .backup is the right choice or if it matters at all.

dirkhh, 2024-04-28:
the number of file name extensions that really are just a zip file...
`.jar` `.apk` `.xlsx` etc etc

wiedehopf, 2024-05-04:
@ramphex 
Can you check that button again and see if it now just works out of the box with the current beta?

ramphex, 2024-05-04:
Updating to b13, once done I'll check

ramphex, 2024-05-04:
Looks good, config and config + graphs download as .backup singular file in Safari

commit fee50297 (2024-04-29): change backup file extension to .backup

Apparently Safari tries to be clever when it downloads .zip files
and that breaks our backup. Great job, Apple.

Fixes: #177
closed 2024-05-04 by wiedehopf