dirkhh/adsb-feeder-image #179: Live Map Configuration URL

askthedads, 2024-04-29:
This is a wonderful image. Great work on the build and ease of setup! I do have a quick question regarding the Live Map view.

Is there any way to disable the ADS-B Feeder Image Configuration header URL in the map view? I couldn't seem to find any configuration parameters to control this. I want to host this externally but prevent viewers from accessing the configuration itself. Apologies in advanced if this is not the appropriate method of inquiring. I did not see any discussion sections in the repo. Thank you!

[image]

dirkhh, 2024-04-29:
there's a link to both Discord and Zulip on the website (and to Zulip on every page of the UI), but this works.

That's a good point. I'll add this in a future version. Thanks for pointing it out.
(having said that - make sure that if you give access to the map to people you only give them access to port 8080 - otherwise all they need to do is switch to port 80...)

askthedads, 2024-04-29:
Thanks @dirkhh ill dive into the discord!

commit 6391ca9b (2024-04-29): add option to disable the config link on map

This simply removes the UI element, it doesn't solve the issue that a user to
whom you make the map available might just change the port used in their web
browser and access the feeder home page that way.

Fixes: #179
closed 2024-05-04 by dirkhh