dirkhh/adsb-feeder-image #181: 2.0.1b4 to 2.0.1b6 failure to update

ramphex, 2024-04-30:
Current:v2.0.1-beta.4(beta)
Board:Raspberry Pi 3 Model B Rev 1.2
Base:adsb-im-raspberrypi64-v1.0.0.img
Kernel:Linux adsb-feeder 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
Containers:
ghcr.io/sdr-enthusiasts/docker-adsb-ultrafeeder:latest-build-346
ghcr.io/sdr-enthusiasts/docker-flightradar24:latest-build-653
ghcr.io/sdr-enthusiasts/docker-piaware:latest-build-474
ghcr.io/sdr-enthusiasts/docker-radarbox:latest-build-629
SDR(s):
SDR(type: 'rtlsdr' address: '001:005', serial: '978')
SDR(type: 'rtlsdr' address: '001:006', serial: '1090')
Ultrafeeder args:
Env variables:
Memory:
               total        used        free      shared  buff/cache   available
Mem:           909Mi       508Mi        38Mi        80Mi       361Mi       256Mi
Swap:           99Mi        95Mi       4.0Mi
Storage:
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       7.1G  4.5G  2.3G  67% /
devtmpfs        325M     0  325M   0% /dev
tmpfs           455M     0  455M   0% /dev/shm
tmpfs           182M   59M  124M  33% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
/dev/mmcblk0p1  255M   32M  224M  13% /boot

When pressing the update (beta) button to update to beta 6, the typical screen comes up saying to stand by as it’s updating. After about 30 seconds, the page refreshes back to the main screen and the version is still beta 4.

dirkhh, 2024-04-30:
That's odd.
Can you try again, and right after it went back to the main screen, use the System->Share Diagnostics feature and post the link here (or send it to me in email to keep it less public - dirkhh@adsb.im)

ramphex, 2024-05-15:
Same issue again going from 2.0.1 to 2.0.2. Emailing you the link to the log.

dirkhh, 2024-05-15:
Well, my guess is undervoltage or bad SD card:
```
/opt/adsb/feeder-update: line 136: 2440418 Segmentation fault      git clone https://github.com/dirkhh/adsb-feeder-image
git clone failed, giving up
Terminated
```
Trying to run `git` to update the feeder image crashes.
Not much that I can do about that.

ramphex, 2024-05-15:
That is strange because the last time this was happening, it was on a RPi3b w/8GB generic card and now it's the RPi5 w/64GB good quality card. I just recently upgraded my setup and actually ended up setting it up from scratch instead of restoring any back ups to avoid any possible conflicts.

What else could be causing this?

wiedehopf, 2024-05-15:
System -> Support info
Base / kernel please.

Suppose it's possible that it's a pi5 issue but when you wrote your previous post we've had to assume pi3 :)

wiedehopf, 2024-05-15:
If you try the update again, does it show the same error in the logs?

You could also try to ssh into the box, then do
```
apt update
apt install --reinstall git
```

Then try the update again.

ramphex, 2024-05-15:
> System -> Support info Base / kernel please.
> 
> Suppose it's possible that it's a pi5 issue but when you wrote your previous post we've had to assume pi3 :)

The initial post issue was on RPi3b, this is the same issue but now on RPi5 with a fresh install.

> Base:adsb-im-raspberrypi64-pi-2-3-4-5-v2.0.1.img
> Kernel:Linux adsb-feeder 6.6.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux

ramphex, 2024-05-15:
> If you try the update again, does it show the same error in the logs?
> 
> You could also try to ssh into the box, then do
> 
> ```
> apt update
> apt install --reinstall git
> ```
> 
> Then try the update again.

Just ran the update again, same result, hangs on the "restarting" page for about 10 secs and refreshes back to the 2.0.1 screen. I'll try to reinstall git

ramphex, 2024-05-15:
Confirming that reinstalling git did solve the issue. Successfully updated to 2.0.2 after.

dirkhh, 2024-05-15:
How you would end up with a git binary that crashes is... rather strange to me. But I'm glad we at least have a workaround identified.

dirkhh, 2024-05-15:
The more I hear about this, the more I think it's a defective SD card

closed 2024-05-15 by dirkhh