dirkhh/adsb-feeder-image #214: Feature Request: SMTP notifications

ramphex, 2024-05-19:
Is there any chance to have stage2 send out SMTP notifications via third party email account to notify of microfeeders reaching high temps, going offline, etc? I use this method of notifications through PowerPanel and NextCloud and love the ability to have it email my phone number so I get it as a text message to notify me of anything pertinent.

dirkhh, 2024-05-19:
Longer term, sure, that would be a module that could be added. It's not on my current to do  list.

Patches are always welcome 😁

CAP-CalebNewville, 2024-11-03:
Maybe a more specialized tool would be better for this - I use [Uptime Kuma](https://github.com/louislam/uptime-kuma) in a VM on my Stage 2 and have it ping the ZeroTier IPs of the micro feeders. It also supports "90+" notification services, besides just SMTP.

dirkhh, 2024-11-04:
I keep thinking that we are getting into the weeds here. This is extremely specific and there are so many different solutions, so many integrations that we'd have to maintain. I'm not convinced there is an obvious, generic, scalable solution that we could support.

dirkhh, 2024-12-28:
as I am cleaning up old issues. this falls squarely into the "not the target audience" category.
Yes, this is useful. No, there's no way to implement something that is generic enough to be useful to many people (mind you, stage 2 is only used by a very small group of people in the first place.

closed 2024-12-28 by dirkhh

ramphex, 2024-12-28:
> Maybe a more specialized tool would be better for this - I use [Uptime Kuma](https://github.com/louislam/uptime-kuma) in a VM on my Stage 2 and have it ping the ZeroTier IPs of the micro feeders. It also supports "90+" notification services, besides just SMTP.

That just checks whether something is actually up, doesn’t notify you of microfeeders going down, sustained high CPU loads, high temps, low disk space, etc