dirkhh/adsb-feeder-image #216: Feature Request: on-device wifi configuration

Ottergoose, 2024-05-22:
Proposed workflow:

- Put SD card that already has ADSB.im image written to it into an SBC
- Wait a moment for boot
- With user's computer or mobile device, connect to "ADSB Feeder" wifi network (which is broadcasting from SBC)
- After connecting, go to http://adsb-feeder.local/
- Detects you're connected directly to wifi from device, fowards you to new wifi configuration page
- Fill out a form with a dropdown of detected networks, or a text box for non-broadcast network ID, and network password, and test connection to confirm it works (possibly show the IP number assigned by new network?)
- If connection works, and confirmed by user, drop the self-broadcast wifi network and instruct user to connect via other network

_**But why do this when the Raspberry Pi Imager makes it so easy to configure WiFi now?**_

- Users cannot reliably self-report network names/passwords, so writing card before arriving on-site to install elsewhere doesn't always work (ask me how I know)
- It's possible to fat-finger credentials when writing configuration to the card (ask me how I know)
- It would enable ADSB.im to be sent preinstalled on a card to less tech-savvy users (friends hosting at a remote location, or even as part of a pre-built kit - you don't need to distribute such a kit, but there seem to be folks out there who might)
- It would allow me to bring an SD card or two with the image installed on it when I deploy a new feeder, and leave my laptop and card reader at home
- It would achieve feature parity of the ADSBx image, which I believe is the only other software out there that's about as easy to configure for multiple feeders (at least for FR24, FlightAware, and ADSBx) as ADSB.im for a newbie

Thanks for your consideration!

dirkhh, 2024-05-22:
while different in some details, I have been working on something like this for a few days now.

Ottergoose, 2024-05-22:
That's great news, thank you! Can't wait to test it out!

dirkhh, 2024-06-01:
I believe that's completed

closed 2024-06-01 by dirkhh