dirkhh/adsb-feeder-image #231: No connection via Port 80 after updating to 2.1.2

c0by85, 2024-06-19:
Port :80 for the webgui is missing / not starting.

```
docker ps -a
CONTAINER ID   IMAGE                                                              COMMAND                  CREATED       STATUS              PORTS                                                                                                                                                                                                                                                                                                                                                                             NAMES
ab0ab3587a9b   ghcr.io/sdr-enthusiasts/docker-piaware:latest-build-498            "/init"                  2 hours ago   Up About a minute   30003/tcp, 30005/tcp, 30105/tcp, 30978-30979/tcp, 0.0.0.0:1093->80/tcp, :::1093->80/tcp, 0.0.0.0:1092->8080/tcp, :::1092->8080/tcp                                                                                                                                                                                                                                                piaware
b894434514d7   amir20/dozzle:latest                                               "/dozzle"                2 hours ago   Up About a minute   0.0.0.0:1094->8080/tcp, :::1094->8080/tcp                                                                                                                                                                                                                                                                                                                                         dozzle
3071aa0c6bb9   alpine                                                             "tail -n0 -f /opt/ad…"   2 hours ago   Up About a minute                                                                                                                                                                                                                                                                                                                                                                                     adsb-setup-proxy
0cf5d314e487   ghcr.io/sdr-enthusiasts/docker-adsb-ultrafeeder:latest-build-410   "/init"                  2 hours ago   Up About a minute   0.0.0.0:30001-30006->30001-30006/tcp, :::30001-30006->30001-30006/tcp, 0.0.0.0:30047->30047/tcp, :::30047->30047/tcp, 0.0.0.0:30104->30104/tcp, :::30104->30104/tcp, 0.0.0.0:30152->30152/tcp, :::30152->30152/tcp, 0.0.0.0:31003-31006->31003-31006/tcp, :::31003-31006->31003-31006/tcp, 0.0.0.0:32006->32006/tcp, :::32006->32006/tcp, 0.0.0.0:1090->80/tcp, :::1090->80/tcp   ultrafeeder
7cabe32a6f45   ghcr.io/sdr-enthusiasts/docker-planefinder:latest-build-371        "/init"                  6 days ago    Up About a minute   0.0.0.0:30053->30053/tcp, :::30053->30053/tcp, 30054/tcp                                                                                                                                                                                                                                                                                                                          pfclient
f90166a75cb5   ghcr.io/sdr-enthusiasts/docker-radarvirtuel:latest-build-574       "/init"                  6 days ago    Up About a minute                                                                                                                                                                                                                                                                                                                                                                                     radarvirtuel
882254e8f964   ghcr.io/sdr-enthusiasts/docker-flightradar24:latest-build-679      "/init"                  6 days ago    Up About a minute   30003/tcp, 0.0.0.0:8754->8754/tcp, :::8754->8754/tcp, 30334/tcp                                                                                                                                                                                                                                                                                                                   fr24feed
```

Via :1090 i can reach the live map.

[image]

```
root@feeder-test:~# docker -v
Docker version 26.1.4, build 5650f9b

root@feeder-test:~# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 23.10
Release:        23.10
Codename:       mantic
```

c0by85, 2024-06-19:
why is it running in dev mode?

```
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on all addresses (0.0.0.0)
 * Running on http://127.0.0.1:1099
 * Running on http://192.168.1.220:1099
```

```
2024-06-19T12:34:39.236Z config.json write: Startup
2024-06-19T12:34:39.241Z config.json write: AF_WEBPORT = 1099
2024-06-19T12:34:39.244Z config.json write: FEEDER_TAR1090_IMAGE_CONFIG_LINK = http://HOSTNAME:1099/
2024-06-19T12:34:39.247Z config.json write: AF_TAR1090_PORT = 1090
2024-06-19T12:34:39.251Z config.json write: AF_UAT978_PORT = 1091
2024-06-19T12:34:39.255Z config.json write: AF_PIAWAREMAP_PORT = 1092
2024-06-19T12:34:39.259Z config.json write: AF_PIAWARESTAT_PORT = 1093
2024-06-19T12:34:39.262Z config.json write: AF_DAZZLE_PORT = 1094
2024-06-19T12:34:39.265Z config.json write: AF_APP_INIT_DONE = True
2024-06-19T12:34:39.276Z config.json write: AF_TAR1090_PORT_ADJUSTED = 1090
```

wiedehopf, 2024-06-19:
That's just a blanket warning from the webserver being used.

App mode was never supposed to be on port 80, you've already quoted the ports it's running on.

c0by85, 2024-06-19:
My PI's running on Port :80.

[image]

wiedehopf, 2024-06-19:
It all depends if it's an image install or an app install.

c0by85, 2024-06-19:
Both were installed with the following command.

`curl https://raw.githubusercontent.com/dirkhh/adsb-feeder-image/main/src/tools/app-install.sh | sudo bash`

wiedehopf, 2024-06-19:
Yep updating will change the port as well.

dirkhh, 2024-06-19:
App installs have the web UI on port 1099 and the map on port 1090
There was a bug that for a while this wasn't handled correctly and the default image ports were used in app mode as well, but this has been fixed in the last release.

closed 2024-06-19 by dirkhh