dirkhh/adsb-feeder-image #235: Raspberry Pi Model 4b is not running with the Current Version

KranzKrone, 2024-06-29:
Hi,

i use a RPI Model 4b with 8Gb RAM.

Step to reproduce:

1. Take a new image of debian 

```
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

3. install all updates
4. install docker docker-compose (Docker version 20.10.24+dfsg1, build 297e128)
5. install ADSB.im cia curl (curl https://raw.githubusercontent.com/dirkhh/adsb-feeder-image/main/src/tools/app-install.sh | sudo bash)

a month ago it worked fine on raspberry pi and manual installation

wiedehopf, 2024-06-29:
The app install runs on port :1099
http://192.168.1.34:1090

Previously due to a bug the app would run on port 80 like the image.

If that's not the issue, the log sanitizer will provide you with a link with details about the system and the logs.
```
bash /opt/adsb/log-sanitizer.sh | curl -F'expires=168' -F'file=@-'  https://0x0.st
```

KranzKrone, 2024-06-29:
Thank  @wiedehopf, i like prefer port 80.

thanks to logs:

Starting dozzle           ... done
2024-06-29T09:01:27.342Z starting AdsbIm.__init__
2024-06-29T09:01:27.408Z config.json write: Startup
2024-06-29T09:01:27.425Z config.json write: _ADSBIM_STATE_LAST_DNS_CHECK = True
2024-06-29T09:01:27.452Z in the end we have
2024-06-29T09:01:27.452Z 1090serial 
2024-06-29T09:01:27.452Z 978serial 
2024-06-29T09:01:27.452Z airspy container is False
2024-06-29T09:01:27.452Z SDRplay container is False
2024-06-29T09:01:27.453Z dump978 container False
2024-06-29T09:01:27.453Z ultrafeeder_config 0
2024-06-29T09:01:27.453Z generating netconfigs for Ultrafeeder
2024-06-29T09:01:27.454Z enabled_aggregators for 0 with agg_sel  and stage2 False
2024-06-29T09:01:27.458Z ended up with Ultrafeeder args []
2024-06-29T09:01:27.458Z base_is_configured: Env(FEEDER_ALT_M, ['']) isn't set up yet
 * Serving Flask app 'app'
 * Debug mode: off
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on all addresses (0.0.0.0)
 * Running on http://127.0.0.1:1099
 * Running on http://192.168.2.189:1099
Press CTRL+C to quit
2024-06-29T09:02:13.413Z director redirecting to setup, base_config not completed
192.168.2.189 - - [29/Jun/2024 11:02:13] "GET / HTTP/1.1" 200 -

closed 2024-06-29 by KranzKrone