dirkhh/adsb-feeder-image #237: Proxmox ISO VM install via GUI (the easy way)

petruut, 2024-07-05:
Before I suggest anything, I want to say this is a beautiful project and I love it!!

is it possible to provide an easy to use ISO for Proxmox? Especially since USB pass through is possible via GUI in a VM, so a user could just do the whole thing via the Proxmox webpage.

the current setup is no problem for experienced users, I’m sure, but as a beginner I struggle.

(what is a beginner doing with prox? Well I think there are many who run some standard LXC or docker VMs on a mini PC in a corner, without much technical knowledge - simplifying this process would be really useful)

dirkhh, 2024-07-05:
I'm reasonably certain that the ISO that is there should work with Proxmox, as long as you make sure the VM you create is set up for BIOS and SATA. I haven't tried it myself, but I've used the ISO with straight kvm setups / via cockpit several times without an issue.
If this doesn't work I'd love to understand why...

petruut, 2024-07-06:
I'm grateful for the quick reply. In releases, there are 2x images labelled with Proxmox. I want to create a VM with one of these, since I need to pass through the USB, and I know that it's a lot easier to do that through the Proxmox GUI for VMs.

When I try to add the ISO (which you have to do ahead of time with Proxmox, before you can build a VM on it), I get this error:

[image]

dirkhh, 2024-07-06:
To reduce download size the ISO is compressed. Please use something like xz or WinZip to decompress it and then things will work fine.

closed 2024-07-09 by dirkhh