dirkhh/adsb-feeder-image #239: DietPi install - interface (and potentially software) not available

petruut, 2024-07-06:
Hi there - great job with this again.

Installed it into a Prox VM of DietPi and downloaded the software via the diet pi-software menu (fresh install).

I can ping the DietPi just fine on its local address, but the :1099 gives me nothing.

How can I troubleshoot this from root@DietPi?

wiedehopf, 2024-07-06:
```
bash /opt/adsb/log-sanitizer.sh | curl -F'expires=24' -F'file=@-'  https://0x0.st
```

This will upload some diagnostic information, you can share the link here.

petruut, 2024-07-06:
Some more information from the DietPi-Services:

[image]

[image]

petruut, 2024-07-06:
[image]

wiedehopf, 2024-07-06:
```
apt install -y jq
```
Then run the log thing again.

petruut, 2024-07-06:
done @wiedehopf 

[image]

wiedehopf, 2024-07-06:
Yeah that wasn't quite needed after all, check my last post again, changed it.

petruut, 2024-07-06:
thanks for that - did it and now we're at: https://0x0.st/XBMQ.txt

the webpage is still not running (gave it a reboot for good measure, but still nothing on 1099)

wiedehopf, 2024-07-06:
Running on port 80, not sure why.

wiedehopf, 2024-07-06:
```
 ls /opt/adsb/*.feeder.image
```

wiedehopf, 2024-07-06:
You'd have to check for example with htop what is using the memory.

You can probably limit the VM to 1.5G memory without issue.

linux will use free memory for file caching, doesn't necessarily mean that it actually needs that much memory.

closed 2024-08-02 by wiedehopf