dirkhh/adsb-feeder-image #242: Boot media for orange pi 5 hard coded to /dev/mmcblk0p1

kelemvor4, 2024-08-18:
The boot media for the orange pi 5 is hard coded to use the optional MMC storage device. In my case, I didn't buy the slower MMC. I use the faster nvme device.  It's worth noting that neither of these storage devices come standard with the orange pi, although htey can be purchased together with it in a bundle.   

Normally the nvme device mounts as /dev/nvme0n1p1 in ubuntu desktop on the OP5, but I suppose there could be a different partition configuration if someone is using a modified rockchip distribution.

Video showing the boot failure: https://www.youtube.com/watch?v=7--Kelf_7Ug

wiedehopf, 2024-08-18:
Not sure how you would work around this.

Possibly partition UUID ... not sure.

Regardless, this seems sufficiently non-standard that it's probably better to just install some debian / ubuntu and run the adsb.im script install on top of that.

dirkhh, 2024-08-18:
We don't have a full image for the Orange Pi 5, only for the 5 plus
Since I don't have access to any of the other flavors of the board, I don't know if and with which limitations the image would work on them.

Certainly there is no intention to use anything other than the SD card out of the box. 

I think the recommendation to install either upstream DietPi or another Debian style distribution and then install the feeder image on top of that seems like your best option.

kelemvor4, 2024-08-18:
I'm using the 5 plus, sorry for not being complete in my description.

The boot device UUID is in the boot image file dietpiEnv.txt

rootdev=UUID=2f7c36a8-4dbd-4019-8795-102c4f747724

Which is CORRECT!  I'm scratching my head as to why it's not respecting the config file.  

This does seem to be a common problem with the Rockchip processors, as I see other users with the same problem (different distributions, however) on Radxa hardware, for example.   Maybe something baked into the firmware?   I've spent too much time on it already…

As you say, I'll give up on this image and manually set things up on Joshua Riek's Ubuntu build for the device.

closed 2024-12-28 by dirkhh