dirkhh/adsb-feeder-image #245: Better support for Mode-S Beast serial output

EvenAR, 2024-10-09:
First of all, thank you for creating this amazing image - it makes sharing data incredibly easy!

I have a Mode-S Beast connected to a Raspberry Pi 4 via USB, and my current setup for using it with adsb.im involves the following steps:

1. Adding the following config to `sudo nano /opt/adsb/config/docker-compose.yml`. This gives the ultrafeeder container access to the Mode-S Beast serial output:
    ```yml
    services:
      ultrafeeder:
        devices:
          - /dev/ttyUSB0:/dev/ttyUSB0
    ```
2. Adding the following environment variable on the "Expert" settings page:
    ```sh
    READSB_DEVICE_TYPE=modesbeast
    ```

.. which took some time to figure out. What are your thoughts on adding this to the default `docker-compose.yml`? It would be even better if "Mode-S Beast over USB" were an option during initial setup :) 

Best regards,
Even

wiedehopf, 2024-10-09:
Very few people with ModeS beast but i'll take a look.

wiedehopf, 2024-10-09:
On the SDR page, can you toggle the lsusb output and post the full output here?

wiedehopf, 2024-10-10:
So i'm not quite sure yet about adding detection logic as this is quite rare.
But we can at least make it unnecessary to edit the yml so only the env var is needed (which persists updates).

```
/opt/adsb/feeder-update origin/dev-wiede
```
could you test if your setup remains working after updating to this dev branch without having to edit the yml?

EvenAR, 2024-10-10:
Removed the config from the yaml, pulled the update, rebooted and it works 👍

wiedehopf, 2024-10-22:
Ok, so with beta and versions following, this will be the only thing required to run a modesbeast:
Add this to environment variables:
```
READSB_DEVICE_TYPE=modesbeast
```

Currently not too eager to automate detection and things because it's a bit of a hassle.
Maybe some day.
Suppose this issue can stay open for other people trying to make it work.

EvenAR, 2024-10-27:
That's fair. Not having to manually edit the configuration files is already a huge improvement. The fact that no one has asked about the Beast before (afaik) probably means there are quite few of us.. Anyways, thanks for your help!

wiedehopf, 2025-05-07:
While this shouldn't happen, i don't think readsb cares about that command line option when using modes beast.
It's using --beast-serial= and if that is not set it uses the default.

You'd have to send me your System -> Share Diagnostics

EvenAR, 2025-05-07:
Here: https://termbin.com/ixyf

wiedehopf, 2025-05-07:
And it was working previously?

Can you run this and see if that fixes it?
```
/opt/adsb/feeder-update origin/dev-wiede
```

I'm not sure why it would but it's worth a try i guess.

EvenAR, 2025-05-10:
I'm not sure why, but after upgrading from v2.2.6 to the stable v2.3.4 it reads data from the Beast as expected. Perhaps my first installation was corrupt 🤔 

Anyways, here are the logs with `READSB_EXTRA_ARGS=--devel=debugSerial`:
[v2.3.4](https://termbin.com/6k96)
[v2.3.5-beta.6](https://termbin.com/mtx5)
[v2.3.5-beta.6(origin/dev-wiede)](https://termbin.com/e1jb)

wiedehopf, 2025-05-10:
Oh i didn't remember how spammy it is.
You should probably remove that env var.

It's possible that there is something that doesn't always work going on.
But that's hard to debug, i'll change readsb to log some startup stuff for ModeS Beast even without that debug option.
That is for the next round of betas so it will be a bit.

FEEDER_READSB_DEVICE is empty so that is all correct.
Modes beasts don't have a serial that's why it says that.
It's expected on the SDR setup page.
It is no longer passed to readsb as --device (it would work like that but it looks weird).

So it's working now on beta?
If you have further issues, maybe open a new issue?
Because it's possible readsb doesn't always work with the ModeS beast, hard to say.

closed 2025-05-10 by wiedehopf

EvenAR, 2025-05-10:
I will, thanks! Yes, it's working with both the beta and the stable version now.

wiedehopf, 2025-05-10:
Seems that i'm defaulting to 3 MegaBaud but the ModeS Beast at least to official specs runs lower transfer rates.

```
READSB_EXTRA_ARGS=--beast-baudrate=921800
READSB_EXTRA_ARGS=--beast-baudrate=1000000
```

You could try those two.
Very possible that this will work with better stability.
Maybe i should just default readsb to 1 MBaud

wiedehopf, 2025-05-12:
https://github.com/wiedehopf/readsb/issues/96#issuecomment-2872434702

Think this might have been related to this.
We'll probably get that change into the next beta.

I don't know why it's working for you without the baudrate being set correctly ... might be a fluke that it's working currently.

EvenAR, 2025-05-14:
I upgraded to v2.3.5-beta.8 and it worked fine. However, once I did a reboot it stopped receiving data from the Beast again.

Logs after reboot: https://termbin.com/dffd

I have now tried several reboots and of my Raspberry Pi 4 without getting the Beast data again. I also tried to recreate containers a couple of times.

wiedehopf, 2025-05-14:
Wait that log has the old readsb version.

.... -.-

Somehow the pinned container doesn't have the current version.

New beta should be available in a moment.

wiedehopf, 2025-05-14:
Try beta 9 please.
The other ModeS beast user reports no issues with that version (he is on bare metal using readsb directly though).

EvenAR, 2025-05-14:
The new beta works 🎉  I also did a reboot and recreated containers without issues

wiedehopf, 2025-05-14:
That's good to hear.
Considering the other user is having no more issues, that would seem to be a definite fix.

Thus adsb.im supports Mode-S Beast out of the box now (on stable once the next stable is out).